mne-tools/mne-python #6597: `mne make_scalp_surfaces` : Surface outer skin  has topological defects

sappelhoff, 2019-07-24:
# Versions
- mne-python master at `d75bce15717ee4cbf2abbdb5c6e5148959000a91` (23rd of July 2019)
- freesurfer `freesurfer-Linux-centos6_x86_64-stable-pub-v6.0.0-2beb96c`

# output of `mne.sys_info()` 
```Text
qt.qpa.screen: QXcbConnection: Could not connect to display 
Could not connect to any X display.
```
:smile: 

I run this on a Linux server via ssh ... but I also get this issue on my own linux machine (Ubuntu 18.04). Perhaps we can add a `try-except` to `mne.sys_info` for these cases.

# How to reproduce

The following commands are run from bash:
```bash
# make sure we have the sample dataset
python -c "import mne; mne.datasets.sample.data_path(verbose=True)"

# make a copy of the freesurfer dir of sample data
# assuming we have our SUBJECTS_DIR in ~/subjects
cp -r ~/mne_data/MNE-sample-data/subjects/sample ~/subjects 

# setup freesurfer
export FREESURFER_HOME=/usr/local/freesurfer
source $FREESURFER_HOME/SetUpFreeSurfer.sh
export SUBJECTS_DIR=~/subjects

# run mne binding for freesurfer
mne make_scalp_surfaces -s sample --overwrite
```

# Error message
```Text
1. Creating a dense scalp tessellation with mkheadsurf...
2. Creating /home/appelhoff/subjects/sample/bem/sample-head-dense.fif ...
outer skin  CM is  -0.03   8.83 -23.48 mm
Surfaces passed the basic topology checks.
3. Creating medium tessellation...
3.1 Decimating the dense tessellation...
3.2 Creating /home/appelhoff/subjects/sample/bem/sample-head-medium.fif
Traceback (most recent call last):
  File "/home/appelhoff/mne-python/mne/commands/mne_make_scalp_surfaces.py", line 136, in <module>
    run()
  File "/home/appelhoff/mne-python/mne/commands/mne_make_scalp_surfaces.py", line 63, in run
    options.no_decimate, options.verbose)
  File "</home/appelhoff/mne-python/mne/externals/decorator.py:decorator-gen-425>", line 2, in _run
  File "/home/appelhoff/mne-python/mne/utils/_logging.py", line 89, in wrapper
    return function(*args, **kwargs)
  File "/home/appelhoff/mne-python/mne/commands/mne_make_scalp_surfaces.py", line 131, in _run
    incomplete=incomplete)
  File "/home/appelhoff/mne-python/mne/bem.py", line 505, in _surfaces_to_bem
    _check_complete_surface(surf, copy=True, incomplete=incomplete)
  File "/home/appelhoff/mne-python/mne/bem.py", line 266, in _check_complete_surface
    raise RuntimeError(msg)
RuntimeError: Surface outer skin  has topological defects: 40 / 30000 vertices have fewer than three neighboring triangles [1079, 1402, 3668, 3786, 6523, 6524, 6598, 6866, 7317, 8067, 8301, 8347, 8422, 8817, 8867, 8940, 8941, 9039, 9197, 9222, 9341, 9433, 10947, 11039, 11163, 11622, 11725, 11880, 11952, 12030, 12420, 12783, 12897, 13225, 13528, 13737, 13937, 13995, 13996, 14305]
```

larsoner, 2019-07-24:
Use `--force`.

We can improve the error message to suggest this as an option if we want.

sappelhoff, 2019-07-24:
Thanks @larsoner that helped. I submitted a PR to improve the error message

PR #6598 (2019-07-25): improve error message make_scalp_surfaces
fixes #6597
closed 2019-07-25 by agramfort