mne-tools/mne-python #6601: GDF events (again!)

papadop, 2019-07-24:
#### Describe the bug
mne-python (0.18.1)
The new _get_annotations_gdf function collapses all the events not found in the global GDF_EVENTS_LUT to the description 'Unknown'. This is done at read_raw_gdf, so all the options in events_from_annotations are useless. 
There is thus no way to recover and distinguish among those 'Unknown' events.

#### Steps and/or code to reproduce
import mne
signal = mne.io.read_raw_gdf('data.gdf',preload=True)
events = mne.events_from_annotations(signal)
print(events)

Of course data.gdf must contain events that are not in GDF_EVENTS_LUT (or some events in GDF_EVENTS_LUT can be removed).

Several tools (e.g. OpenViBE) can produce tags which are not in the list of predefined events depicted in the gdf standard. Collapsing all of those to 'Unknown' is bad because the information lost is unrecoverable.

For my file, the above code gives:
(array([[  5538,      0,      1],
       [  5844,      0,      1],
       [  6176,      0,      1],
       ...,
       [164368,      0,      1],
       [164609,      0,      1],
       [164768,      0,      1]]), {'Unknown': 1, 'condition 11': 2, 'condition 12': 3, 'condition 14': 4, 'condition 17': 5, 'condition 2': 6, 'condition 20': 7, 'condition 23': 8, 'condition 3': 9, 'condition 48': 10})

With an old version of mne-python (0.13.1) and using find_edf_events, we get:
>>> events
[3049, array([  5538,   5844,   6176, ..., 164368, 164609, 164768], dtype=uint32), array([181, 193, 190, ..., 199, 190, 182], dtype=uint16), array([0, 0, 0, ..., 0, 0, 0], dtype=int32), array([1, 1, 1, ..., 1, 1, 1], dtype=uint32)]
>>> np.unique(events[2])
array([  0,   2,   3,  11,  12,  14,  17,  20,  23,  27,  36,  37,  39,
        40,  42,  44,  45,  46,  48, 181, 182, 183, 184, 187, 190, 193,
       195, 198, 199], dtype=uint16)

Events 2,   3,  11,  12,  14,  17,  20,  23, 48 are kept but all the others are collapses under 'Unknown': 1

#### Expected results
Identity of initial events should be recoverable.... Actually, I must say that I do not understand why events are re-numbered by default through the sorting of descriptions at gdf read time... This seems unnecessary (events are already numbered) and unreliable (imagine the case of a name change in gdf, or the addition of an event between two similar experiments, besides the currently described case of non standardized user defined events).
I would expect read_raw_gdf to do minimal changes to this event table and the renaming to be done at events_from_annotations where the user has some leverage of what happens using the event_id parameter.

On the same trend, _get_annotations_gdf converts gdf time stamps to seconds which are then converted back to integers in events_from_annotations. This will probably induces some interesting rounding effects... Not a big deal, but not totally reliable/reproducible....

#### Actual results
See above

#### Additional information
>>> mne.sys_info()
/usr/lib64/python3.7/site-packages/sklearn/externals/six.py:7: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
/usr/lib64/python3.7/site-packages/sklearn/utils/__init__.py:4: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  from collections import Sequence
Platform:      Linux-5.1.17-300.fc30.x86_64-x86_64-with-fedora-30-Thirty
Python:        3.7.3 (default, May 11 2019, 00:38:04)  [GCC 9.1.1 20190503 (Red Hat 9.1.1-1)]
Executable:    /usr/bin/python3
CPU:           x86_64: 12 cores
Memory:        31.1 GB

mne:           0.18.1
numpy:         1.16.4 {blas=openblasp, lapack=openblasp}
scipy:         1.2.0
matplotlib:    3.0.3 {backend=Qt5Agg}

sklearn:       0.19.1
nibabel:       2.4.1
mayavi:        Not found
cupy:          Not found
pandas:        0.23.4
dipy:          Not found

papadop, 2019-07-24:
This simple patch is a stop gap, it keeps the event id in the descriptions and put the LUT as a static variable of RawGDF to allow the user to manipulate the table before reading a file... 
As depicted above, I do not think this is a reliable solution in the long term...

mururoa-> git diff --cached origin/master
diff --git a/mne/io/edf/edf.py b/mne/io/edf/edf.py
index 6061cc817..aee293f75 100644
--- a/mne/io/edf/edf.py
+++ b/mne/io/edf/edf.py
@@ -28,9 +28,6 @@ from ...annotations import Annotations, events_from_annotations
 from ._utils import _load_gdf_events_lut
 
 
-GDF_EVENTS_LUT = _load_gdf_events_lut()
-
-
 @deprecated('find_edf_events is deprecated in 0.18, and will be removed'
             ' in 0.19. Please use `mne.events_from_annotations` instead')
 def find_edf_events(raw):
@@ -239,6 +236,8 @@ class RawGDF(BaseRaw):
     mne.io.read_raw_gdf : Recommended way to read GDF files.
     """
 
+    GDF_EVENTS_LUT = _load_gdf_events_lut()
+
     @verbose
     def __init__(self, input_fname, montage, eog=None, misc=None,
                  stim_channel='auto', exclude=(), preload=False, verbose=None):
@@ -277,8 +276,7 @@ class RawGDF(BaseRaw):
                 ' in 0.19. Please use `mne.events_from_annotations` instead')
     def find_edf_events(self):
         return events_from_annotations(self)
-
-
+        
 def _read_ch(fid, subtype, samp, dtype_byte, dtype=None):
     """Read a number of samples for a single channel."""
     # BDF
@@ -1437,8 +1435,7 @@ def _get_annotations_gdf(edf_info, sfreq):
     if events is not None and events[1].shape[0] > 0:
         onset = events[1] / sfreq
         duration = events[4] / sfreq
-        desc = [GDF_EVENTS_LUT[key]
-                if key in GDF_EVENTS_LUT else 'Unknown'
+        desc = [RawGDF.GDF_EVENTS_LUT[key]
+                if key in RawGDF.GDF_EVENTS_LUT else 'Undefined('+key+')'
                 for key in events[2]]
-
     return onset, duration, desc

papadop, 2019-07-24:
And here is a datafile.
[data.zip](https://github.com/mne-tools/mne-python/files/3427830/data.zip)

agramfort, 2019-07-24:
Can you send us a PR so we can review the diff more easily? Thanks

papadop, 2019-07-24:
Sure. I will do it in a second. But again, I think this is just a temporary fix...

PR #6602 (2019-08-02): ENH: make GDF unknown events codes visible
#### Reference issue
Fix #6601.

#### What does this implement/fix?
This makes two things:
- The 'Unknown' description is replaced by 'Undefined(number)' where number is the number in the gdf event table. This avoids collapsing all 'Unknown' events under the same tag and allows to keep event identity.
- The global variable  GDF_EVENTS_LUT becomes a RawGDF static member. This at least allows the user to add its own events to the table before reading a file.

Example:
mne.io.edf.edf.RawGDF.GDF_EVENTS_LUT[198] = 'Non Target'
mne.io.edf.edf.RawGDF.GDF_EVE [... cut]
closed 2019-08-02 by massich