mne-tools/mne-python #6610: Allow opening EEGLAB .set files without any events

alamkanak, 2019-07-29:
It is obvious that when you have an epoched `.set` file exported from eeglab with no events in it, mne fails to load it with `read_epochs_eeglab` or `read_raw_eeglab`. When trying to open with `read_epochs_eeglab`, it throws the error: 

> IndexError: list index out of range

I also tried to open it by setting `events` parameter to empty array and `None` but it still failed. And when trying to open with `read_raw_eeglab`, it throws the error:

> TypeError: The number of trials is 43. It must be 1 for raw files. Please use `mne.io.read_epochs_eeglab` if the .set file contains epochs.

I really need to open the epoched set file with no events in it.

agramfort, 2019-07-29:
Can you share a file so we can test?

alamkanak, 2019-07-29:
@agramfort Sure thing. Here is one file. https://drive.google.com/open?id=1hhc5VC90M4cDs8vQhWf43KhdGI4bubFx

agramfort, 2019-07-29:
this file is broken:

ipdb> eeg.data
'clean-prestimulus.fdt'

I don't have this file.

make sure you use eeglab save routines to save your files.

>

mmagnuski, 2019-07-29:
@agramfort Eeglab reader should work in this case too, I think (it would work in eeglab from what I remember). The easiest is if you rename both files so that `eeg.data` matches the name of the fdt file.
 
@alamkanak Which mne-python version are you using? There have been problems with no events some time ago but they were fixed. Can you reproduce the problem with mne-python master branch? Could you also paste the full error traceback when trying to open the file with read_epochs_eeglab?

mmagnuski, 2019-07-29:
Ok, I looked at the files. The reading fails on master due to the fact that this is an epochs file without events. Also the `EEG.epoch` field is empty. This is very uncommon, @alamkanak how did you obtain those files?

alamkanak, 2019-07-29:
@mmagnuski Thank you for looking into it. To obtain that file, I epoched the EEG from -1000ms to -100ms in eeglab. So the eeg signals end 100ms before the events. This is because I am only interested in that part of the signals. I had to crop that in eeglab due to a standard pipeline and I must not do that later with mne.

mmagnuski, 2019-07-29:
@alamkanak Thanks, which eeglab version were you using then? I am just wondering if eeglab always drops information about epoching events if they are outside of epoch's range (even in `EEG.epoch`). Maybe you know @jona-sassenhagen?
The problem with such epochs behavior is that because you no longer have any information about the events that were used in epoching you wouldn't be able to compare conditions etc. In fact, mne assumes that you always have event information if you have epoched eeglab data. However if this is standard eeglab behavior (not saving any event information for epoching events outside of epoch range) - then we should probably add support for this.
For now it would be easiest for you @alamkanak if you epoch -1000 to 0 ms for example. Then the event info will be retained and you would be able to read the epochs into mne.

larsoner, 2019-08-20:
I'd be fine with emitting a warning that events information has been lost and still reading the file, @mmagnuski can you make a PR?

mmagnuski, 2019-08-20:
Sure, I can do it this week - would we populate event_id and events and other necessary fields (not sure what other) with some default values then?

larsoner, 2019-08-20:
Yes I would just make `events = np.column_stack((np.arange(len(events)), 0, np.ones(len(events), int)))` and `event_id=dict(unknown=1)`

agramfort, 2019-12-28:
@mmagnuski do you still plan to look into this?

mmagnuski, 2019-12-29:
Yes, sorry, I completely forgot. But I won't be able to do it this year. I can get to it on Friday, though.

agramfort, 2019-12-30:
Whenever you can is good. Thx

guoyongjian111, 2021-11-26:
so i have a rest eeg data(.set), epochs=119,but events = 0  ,what should i do can read this data?please