mne-tools/mne-python #6613: New version of brain vision data

Mehrabkhazraei, 2019-07-29:


teonbrooks, 2019-07-29:
hi @Mehrabkhazraei, ok, I just read the previous issue and have better context for the problem at hand. do you know if Brain Product provides the data specification for the new header and marker file. I don't know what the differences are between the two versions so we're unable to assist without this information. Could you ask them if they have any plans to release these specifications like they did for the VHDR/VMRK? 
I am not actively developing on mne-python, but if you get information, I am sure I can help guide you and whomever wanting to help add this feature

sappelhoff, 2019-08-15:
Hi @Mehrabkhazraei unfortunately I cannot open your example data - the .rar archive seems to be broken.

I agree with @teonbrooks that more information (also from BV customer support) would be nice!

teonbrooks, 2019-08-15:
@sappelhoff do you know anyone at Brain Product who might be able to help? Also I guess this means that we would need to convert all ahdr and associated files to vhdr if someone wanted to make their data compatible

sappelhoff, 2019-08-15:
> do you know anyone at Brain Product who might be able to help? 

I advised @Mehrabkhazraei in #6556 to contact support@brainproducts.com ... I still think that that's the best way forward :-) 

Ideally @Mehrabkhazraei would take over the communication with Brain Products ... and we can assist to implement reading/conversion from VAMP format to BrainVision format in MNE-Python

cbrnr, 2019-08-19:
Several years ago someone asked this question [on the EEGLAB mailing list](https://sccn.ucsd.edu/pipermail/eeglablist/2013/007301.html), and the answer was that the data are encrypted and cannot be loaded (the key is probably the hardware dongle). If this is true, there's nothing we can do other than advise people to use the normal .vhdr/.vmrk/.eeg format.

teonbrooks, 2019-08-30:
I think this then falls into the category of wontfix as we actually can't fix it. closing the issue.

@Mehrabkhazraei is there anyway you can have your data exported to the vhdr/vmrk/eeg format?

closed 2019-08-30 by teonbrooks

berndf, 2021-03-08:
Hello everybody!
I was also desperately looking for information on this V-Amps file format and only found this information about encryption. I checked back with Brain Products and received the information that there is a signature but no encryption of the actual data.
What makes the data look "encrypted" is the fact that there is an additional channel in the binary data (as last channel). So instead of, say, 6 channels described by the header the multiplexed format contains data for a 7th channel.
Note that Arnaud Delorme has introduced this in his repo 9 months ago: https://github.com/arnodelorme/bva-io

sappelhoff, 2021-03-08:
Very interesting, thanks for the info @berndf. Could you by any chance post the correspondence that you had with Brain Products in this issue thread?

I'd be very interested to know why there is a seventh channel ... and what data it contains, and why it is not advertised by the header. Does it mean there is always one more channel than advertised by the header?

Mehrabkhazraei, 2021-03-08:
@sappelhoff indeed the extra channel is nothing than an electrical reference channel. It is not really related to any kind of encryption. @berndf Actually there is no encryption in recorded files. They are just using different type of saving formats. By the way, if you are up to change the format to *.vhdr and *.ahdr, you just need to go to the settings of your recorder app and change the saving formats.

berndf, 2021-03-08:
@sappelhoff Exactly, if the header advertises N channels there are N+1 in the binary data. The "Signature channel" is apparently always the data type of the other channels. The correspondence with BP was in German, but the information given was that the V-Amps were shipped in a bundle with a special version of the Analyzer at a low price, and that the signature was added to render this Analyzer only suitable to analyze the V-Amp data but not other data. So the aim was never to stop other software from processing data acquired with V-Amp. The general Analyzer can always analyze all files (except for a bug that apparently prevented it to last year...)

Lychfindel, 2022-04-12:
Hi all,
I had the same problem with a V-Amp and the BrainVision support told me that indeed the extra channel is a kind of encryption.
Here their reply:

> Thank you for your speedy reply. You are right with n+1 channel :)
> It is a kind of encryption. You can trust the other channels/ units.
> Please find attached written what you already found out.

I have forked and modified the mne code to handle the ahdr file, I'm going to submit a pr with the changes.
P.S. The implementation for the eeglab plugin bva-io was also pull requested by me, and accepted by Arnaud after authorization from BrainVision team.

sappelhoff, 2022-04-13:
xref https://github.com/arnodelorme/bva-io/pull/10

PR #10515 (2022-04-26): Add support for BrainVision ahdr/amrk format, as produced by the V-AMP
#### Reference issue
Example: Fixes #6613

#### What does this implement/fix?
The PR implements the acquisition of BrainVision data in ahdr format. It also adjust the header regular expression to handle ahdr data saved with V-AMP.

#### Additional information
The ahdr format introduces a fake channel in the data as a kind of encryption, the rest of channels and unit can be trusted. Therefore, in the submitted code, the data are read considering one more channel, and after the initialization of the BrainVision object, this c [... cut]