mne-tools/mne-python #6615: ICA.plot_sources(epoch)

jshanna100, 2019-07-30:
The plot_sources method for ICA only produces a wall of blue when the Inst is an Epoch. This occurs even with a fresh conda environment with the stable .18 version. I assume this is a scaling issue, but trying to manually scale it down hasn't gotten me anywhere.

You should be able to reproduce this as follows:
```Python
import mne
path =  mne.datasets.sample.data_path()
raw = mne.io.Raw(path + '/MEG/sample/sample_audvis_raw.fif')
events = mne.read_events(path+"/MEG/sample/sample_audvis_raw-eve.fif")
epo = mne.Epochs(raw,events)
ica = mne.preprocessing.ICA(n_components=100)
ica.fit(epo)
ica.plot_sources(epo)
```

agramfort, 2019-07-30:
I confirm the bug. Can someone have a look?

>

larsoner, 2019-07-30:
Probably an auto-scaling problem. I can take a look

PR #6617 (2019-08-01): MRG, BUG: Fix scaling of sources
Closes #6615.

@jshanna100 can you confirm this fixes your problem?
closed 2019-08-01 by agramfort