mne-tools/mne-python #6616: Q: default value of baseline in Epochs

milan-rybar, 2019-07-30:
What is the motivation for the default value of `baseline=(None, 0)` in `Epochs` constructor? This subtracts the value of the first time point in each epoch.

I can imagine mainly for visualization purposes. However, this can be quite problematic for analysis if one doesn't know about this default behaviour. This was my case. I assumed it uses "raw" data and my whole analysis is based on it, which is now wrong.
From the API perspective, I would assume there is no baseline correction unless specified.

agramfort, 2019-07-30:
put baseline=None then

baseline defaults to (None, 0) as if tmin is negative in ERP/ERF study
is subtracts the mean during baseline

>

larsoner, 2019-07-30:
The `tmin=0` case is common enough that it might warrant a warning. People will probably never actually want a one-sample baseline like this. And if they do we can in the warning tell them to use `baseline=(0., 0.)` to avoid the warning.

agramfort, 2019-07-30:
shall we raise an error with tmin=0 and baseline=(None, 0) ?

larsoner, 2019-07-30:
Sure, that's probably even better than a warning. @milan-rybar do you want to try a PR for this?

milan-rybar, 2019-07-30:
> baseline defaults to (None, 0) as if tmin is negative in ERP/ERF study is subtracts the mean during baseline

I don't think about MNE as a tool primary for ERP/ERF study. Thus, I believe default values shouldn't be based on one particular case of all possible analysis. It would make more sense to remove default values for `tmin`, `tmax`, and set `baseline=None`. This way, the user is directly forced to provide desired values and it doesn't do anything unexpected in the background.

> Sure, that's probably even better than a warning. @milan-rybar do you want to try a PR for this?

I agree with rasing an error for the combination of `tmin=0` and `baseline=(None, 0)`. I can do PR for this.

agramfort, 2019-07-30:
problem is backward compatibility. I would break a lot of code if we change
this so the deprecation cost is to be considered.

>

PR #6650 (2019-08-13): FIX: Raise exception for default value of baseline and `tmin=0` in `Epochs`
Closes #6616

Raise an exception when using the default value of `baseline=(None, 0)` with `tmin=0` in `Epochs`. The result would be one-sample baseline which is not desired in most cases. One should use `baseline=(0, 0)` if this is desired.
closed 2019-08-13 by larsoner