mne-tools/mne-python #6618: tight_layout vertically compresses PSD plots

jshanna100, 2019-07-31:
The extension of spatial colors from Raw to Epochs in #6451 has revealed a bug with the use of tight_layout.

https://github.com/mne-tools/mne-python/blob/cc836c2635b7a720ec6c60074a1997fe52d7ca2f/mne/viz/utils.py#L3064-L3066

Tight layout often (but not always!) vertically compresses the PSD plots when spatial_colors are used with Epochs. I thought at first this had something to do with the changes in #6451, but it turns out to have always been there. It just wasn't apparent before because it only happens when the channels are relatively flat and uniform - which can sometimes happen with the multitaper psd method used by method Epochs.plot_psd, and also because because tight_layout was never used in the average=True case, which up until #6451 was the only option available for Epochs.plot_psd.

You can test that this is the case by taking the latest stable release and producing a vertically compressed PSD plot even for Raw.plot_psd, but setting the n_fft variable to an extremely low value - thus simulating the very broad window used in Epochs with multitaper fft, and the resultant flat, uniform PSDs.

What would then be the sensible thing to do here? Is simply removing the call to tight_layout an option? It was after all never used for average=True, and that seems fine?

agramfort, 2019-07-31:
please share a tiny script to replicate and share a picture of what looks
wrong. thx

>

jshanna100, 2019-07-31:
For some reason, I cannot replicate the bug with the sample data, but here are some pictures of some Raw data from our lab, with a Magnes 3600 248 magnetometer system.

First, normal:
`raw.plot_psd()`
[image]
Now, compressed.
`raw.plot_psd(n_fft=50)`
[image]
I don't know if it's clear from the picture above, but the axis is significantly shorter than the figure.

Switching to development version, and Epochs.plot_psd(average=False), it can in some cases be really severe.
[image]
As I mentioned above, it's somehow fine for the sample/audvis...
```
path =  mne.datasets.sample.data_path()
raw = mne.io.Raw(path + '/MEG/sample/sample_audvis_raw.fif')
picks = mne.pick_types(raw.info,meg="mag")
raw.plot_psd(picks=picks,n_fft=50)
```
[image]
Not sure what to make of that...

jshanna100, 2019-07-31:
In any case, removing tight_layout fixes all these problems, though I don't know whether that breaks something somewhere else.

larsoner, 2019-07-31:
`tight_layout` problems are usually indicative that something wrong is being done in the plot. Or possible there is some `mpl` bug. I don't want to get rid of it because in general it makes things cleaner.

Can you see if `subplots_adjust` works better for your data and still works for `sample`?

jshanna100, 2019-08-01:
It looks good on my data, but with multiple channel type psd display for the sample data the subplot titles overlap onto the x-axis labels of the image above them, unless you increase the size of the figure.

larsoner, 2019-08-01:
`subplots_adjust` has some parameters you can probably tweak to get both to look good. Did you try? Probably `hspace` is what you'd need here

jshanna100, 2019-08-01:
I did try several values between 0.1 and 0.4 and didn't have any luck, but I'll experiment a bit more with it tomorrow.

jshanna100, 2019-08-02:
hspace=0.5 appears to do it.

larsoner, 2019-08-02:
+1 for PR to change it. In your PR, can you make some minor change to an example that uses `raw.plot_psd` (and/or `epochs.plot_psd`)? It will cause CircleCI to render it for us so we can easily see the output

PR #6625 (2019-08-05): Epobutt bugfix
Closes #6618

The first figure in examples/plot_compute_raw_data_spectrum now uses average=False.
closed 2019-08-05 by massich