mne-tools/mne-python #6621: [BUG] set_annotations fails for data with a meas_date prior to 1970

sappelhoff, 2019-08-02:
Try reading this [example_file](https://drive.google.com/file/d/1vZCW3wUQ_0gemac_NFO9LtYVlG70rt-R/view?usp=sharing) (800MB) and check out `raw.info['meas_date']` --> it will be negative.

That's because the data has been anonymized (cc @agramfort ) to some time prior 1970.

That's an issue, because we represent `raw.info['meas_date']` in unix time, which is 0 at 1970, and thus negative for all prior years (strictly speaking, it's even undefined)

A negative `meas_date` will raise an error when make an annotation object from scratch and try to set it with `raw.set_annotations()`.

That is because of this code:

in `set_annotations`: https://github.com/mne-tools/mne-python/blob/368a490f9d7a298050ee04c2aa7882629c9fab21/mne/io/base.py#L792

--> `time_of_first_sample` will be negative if `meas_date` is negative

then later we do: https://github.com/mne-tools/mne-python/blob/368a490f9d7a298050ee04c2aa7882629c9fab21/mne/io/base.py#L798-L801

i.e., pass a negative `tmin` to the `crop` function, which then raises an error: https://github.com/mne-tools/mne-python/blob/368a490f9d7a298050ee04c2aa7882629c9fab21/mne/annotations.py#L363-L364

agramfort, 2019-08-02:
If there is no limitations with fif files to have negative time stamps we should handle these files smoothly

larsoner, 2019-08-02:
Yes this seems like a bug in our Annotations handling somewhere. @massich can you look?

It should not matter what `meas_date` relative to 1970 is (negative or positive)

massich, 2019-08-02:
yes I was just looking at it

massich, 2019-08-05:
What was the logic behind `DATE_NONE` and anonymization? 

I understand that is a convention to move the date to 1930 or so for anonymization purposes. But do people usually keep the rest of the information i.e: imagine that I have one record every month/year for a particular subject. Do they only offset to 1/1/1930 and keep all the recordings with respect to that date or do we fold all recordings to the same date?

sappelhoff, 2019-08-05:
@massich see: https://github.com/mne-tools/mne-bids/issues/193#issue-436776387

I don't know about how anonymization is currently handled in MNE, but I think the way we SHOULD handle this is to allow for an anonymization by a certain offset in days ... this would then be applied to all data recordings of a subject, so that the relative time distances keep their logic.

the `anonymization_offset_days` would be set to a sensible default but could also be given as an input to `raw.annonymize(anonymization_offset_days=36500)`

I briefly discussed this with @larsoner at the last MNE sprint, maybe he wants to weigh in

agramfort, 2019-08-05:
yes we need to have such anonymization routines in mne-python

the reason why you cannot set the date to None is that you want to
keep the order between
datasets eg for longitdinal studies etc. So yes we should use random
day offsets and make
sure the dates can be before 1970. The latest mne-c command does this
with .fif files

PR #6628 (2019-08-05): FIX: set_annotations when meas_date is prior to 1970
fixes #6621
closed 2019-08-05 by sappelhoff