mne-tools/mne-python #6627: Unable to run python 3.5 environment locally

massich, 2019-08-03:
I was trying to run the tests for py3.5 env in Travis (`numpy=1.11 scipy=0.17 matplotlib=1.5 pandas=0.18 scikit-learn=0.18 pytest pytest-cov pytest-mock pytest-timeout`) and I can't run the testsuit.

Apparently, it broke in #6603. But that's not the case in Travis. I was not able to figure out what was it.

```sh
~/code/mne-python master 16s
(mne) ❯ conda activate mne35

~/code/mne-python master
(mne35) ❯ git checkout master~9
Note: checking out 'master~9'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at f39331cc5 MRG Default ica.plot_properties to a parametric CI (#6270)

~/code/mne-python f39331c...
(mne35) ❯ python -c "import mne; mne.sys_info()"                            
Platform:      Linux-4.15.0-46-generic-x86_64-with-debian-buster-sid
Python:        3.5.5 | packaged by conda-forge | (default, Jul 23 2018, 23:45:43)  [GCC 4.8.2 20140120 (Red Hat 4.8.2-15)]
Executable:    /home/sik/miniconda3/envs/mne35/bin/python
CPU:           x86_64: 4 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.19.dev0
numpy:         1.11.2 {lapack=/home/sik/miniconda3/envs/mne35/lib, blas=/home/sik/miniconda3/envs/mne35/lib}
scipy:         0.17.1
matplotlib:    1.5.3 {backend=Qt5Agg}

sklearn:       0.18.1
nibabel:       Not found
mayavi:        Not found
cupy:          Not found
pandas:        0.18.0
dipy:          Not found

~/code/mne-python f39331c...
(mne35) ❯ git checkout master~7                 
Previous HEAD position was f39331cc5 MRG Default ica.plot_properties to a parametric CI (#6270)
HEAD is now at 3acb8dc8c [MRG] Conversion of somato data to BIDS (#6414)

~/code/mne-python 3acb8dc...
(mne35) ❯ python -c "import mne; mne.sys_info()"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/sik/code/mne-python/mne/__init__.py", line 22, in <module>
    from .utils import (set_log_level, set_log_file, verbose, set_config,
  File "/home/sik/code/mne-python/mne/utils/__init__.py", line 23, in <module>
    from .fetching import _fetch_file, _url_to_local_path
  File "/home/sik/code/mne-python/mne/utils/fetching.py", line 14, in <module>
    from .numerics import hashfunc
  File "/home/sik/code/mne-python/mne/utils/numerics.py", line 24, in <module>
    from .linalg import _svd_lwork, _repeated_svd, dgemm, zgemm
  File "/home/sik/code/mne-python/mne/utils/linalg.py", line 44, in <module>
    dgesvd, dgesvd_lwork = linalg.get_lapack_funcs(('gesvd', 'gesvd_lwork'), (_d,))
  File "/home/sik/miniconda3/envs/mne35/lib/python3.5/site-packages/scipy/linalg/lapack.py", line 444, in get_lapack_funcs
    "flapack", "clapack", _lapack_alias)
  File "/home/sik/miniconda3/envs/mne35/lib/python3.5/site-packages/scipy/linalg/blas.py", line 257, in _get_funcs
    '%s function %s could not be found' % (lib_name, func_name))
ValueError: LAPACK function dgesvd could not be found
```

I'm using `scipy-0.17.1` which is the requirement.

Any ideas?

cc: @larsoner

larsoner, 2019-08-20:
Did you run the same commands locally that Travis uses? This should get you the same env, assuming you are also on Linux.

massich, 2019-08-20:
It was a while ago. But there was no change in Travis (that I was aware of). And I could run one commit but not the other.

massich, 2019-09-04:
I'm back into having issues with this. I use a clean conda install with a subset of the minimum dependencies to run mne. I've checked the ci-helpers used in Travis but I couldn't see what I'm doing wrong.

```py
~/code/mne-python read_dig_montage_polhemus* 9s
(mne35) ❯ conda create --name mne35_bis python=3.5 numpy=1.11 scipy=0.17 scikit-learn=0.18 pytest pytest-cov pytest-mock pytest-timeout 
Collecting package metadata (current_repodata.json): done
Solving environment: failed
Collecting package metadata (repodata.json): done
Solving environment: done

==> WARNING: A newer version of conda exists. <==
  current version: 4.7.5
  latest version: 4.7.11

Please update conda by running

    $ conda update -n base conda

## Package Plan ##

  environment location: /home/sik/miniconda3/envs/mne35_bis

  added / updated specs:
    - numpy=1.11
    - pytest
    - pytest-cov
    - pytest-mock
    - pytest-timeout
    - python=3.5
    - scikit-learn=0.18
    - scipy=0.17

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    more-itertools-7.2.0       |             py_0          48 KB  conda-forge
    pluggy-0.11.0              |             py_0          17 KB  conda-forge
    pytest-cov-2.7.1           |             py_0          17 KB  conda-forge
    pytest-mock-1.10.0         |           py35_0          15 KB  conda-forge
    ------------------------------------------------------------
                                           Total:          97 KB

The following NEW packages will be INSTALLED:

  _libgcc_mutex      pkgs/main/linux-64::_libgcc_mutex-0.1-main
  atomicwrites       conda-forge/noarch::atomicwrites-1.3.0-py_0
  attrs              conda-forge/noarch::attrs-19.1.0-py_0
  blas               conda-forge/linux-64::blas-1.1-openblas
  bzip2              conda-forge/linux-64::bzip2-1.0.8-h516909a_0
  ca-certificates    conda-forge/linux-64::ca-certificates-2019.6.16-hecc5488_0
  certifi            conda-forge/linux-64::certifi-2018.8.24-py35_1001
  coverage           conda-forge/linux-64::coverage-4.5.1-py35h470a237_1
  libffi             conda-forge/linux-64::libffi-3.2.1-he1b5a44_1006
  libgcc-ng          pkgs/main/linux-64::libgcc-ng-9.1.0-hdf63c60_0
  libgfortran        conda-forge/linux-64::libgfortran-3.0.0-1
  libstdcxx-ng       pkgs/main/linux-64::libstdcxx-ng-9.1.0-hdf63c60_0
  more-itertools     conda-forge/noarch::more-itertools-7.2.0-py_0
  ncurses            conda-forge/linux-64::ncurses-6.1-hf484d3e_1002
  numpy              conda-forge/linux-64::numpy-1.11.3-py35_blas_openblas_202
  openblas           conda-forge/linux-64::openblas-0.2.19-2
  openssl            conda-forge/linux-64::openssl-1.0.2r-h14c3975_0
  pathlib2           conda-forge/linux-64::pathlib2-2.3.2-py35_0
  pip                conda-forge/linux-64::pip-18.0-py35_1001
  pluggy             conda-forge/noarch::pluggy-0.11.0-py_0
  py                 conda-forge/noarch::py-1.8.0-py_0
  pytest             conda-forge/linux-64::pytest-3.8.1-py35_0
  pytest-cov         conda-forge/noarch::pytest-cov-2.7.1-py_0
  pytest-mock        conda-forge/linux-64::pytest-mock-1.10.0-py35_0
  pytest-timeout     conda-forge/noarch::pytest-timeout-1.3.3-py_0
[... 31 lines cut ...]
~/code/mne-python read_dig_montage_polhemus* 23s
(mne35) ❯ conda activate mne35_bis                                                                                                     

~/code/mne-python read_dig_montage_polhemus*
(mne35_bis) ❯ python -c "import mne; mne.sys_info()"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/sik/code/mne-python/mne/__init__.py", line 22, in <module>
    from .utils import (set_log_level, set_log_file, verbose, set_config,
  File "/home/sik/code/mne-python/mne/utils/__init__.py", line 9, in <module>
    from .check import (check_fname, check_version, check_random_state,
  File "/home/sik/code/mne-python/mne/utils/check.py", line 15, in <module>
    from ._logging import warn, logger
  File "/home/sik/code/mne-python/mne/utils/_logging.py", line 15, in <module>
    from ..fixes import _get_args
  File "/home/sik/code/mne-python/mne/fixes.py", line 22, in <module>
    from scipy import linalg
ModuleNotFoundError: No module named 'scipy'

~/code/mne-python read_dig_montage_polhemus*
(mne35_bis) ❯ python -c "import scipy"              
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'scipy'
```

larsoner, 2019-09-04:
```
larsoner@bunk:~$ export PATH=/home/larsoner/anaconda3/bin:$PATH
larsoner@bunk:~$ conda update -n base conda  # get to 4.7.11
larsoner@bunk:~$ conda create --name mne35_bis python=3.5 numpy=1.11 scipy=0.17 scikit-learn=0.18 pytest pytest-cov pytest-mock pytest-timeout
Collecting package metadata (repodata.json): done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - scipy=0.17
  - scikit-learn=0.18
```
Not sure why it even let you try. I can't seem to make a new 3.5 env with numpy 1.11 and SciPy 0.17, but I have one / test one with slightly newer NumPy / SciPy.

To really solve this you'll need to parse the Travis log:

https://travis-ci.org/mne-tools/mne-python/jobs/580524370

In particular look at the `conda` logs, it does use 4.7.11 and does not end up using as old SciPy/NumPy as we want it to:
```
The following NEW packages will be INSTALLED:
...
  numpy              pkgs/main/linux-64::numpy-1.11.3-py35h1d66e8a_10
  scikit-learn       pkgs/main/linux-64::scikit-learn-0.20.0-py35h4989274_1
  scipy              pkgs/main/linux-64::scipy-1.1.0-py35hd20e5f9_0
...
```
That's a bummer but I'm not sure that there is too much that we can do. For example:
```
(py35) larsoner@bunk:~/python/mne-python$ conda create -n mneold numpy=1.11 scipy=0.17 matplotlib=1.5 pandas=0.18 scikit-learn=0.18 pytest pytest-cov pytest-mock pytest-timeout
Collecting package metadata (repodata.json): done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - matplotlib=1.5
  - scipy=0.17
  - scikit-learn=0.18
  - pandas=0.18
```
Same with `conda-forge`:
```
conda create -c conda-forge -n mneold numpy=1.11 scipy=0.17 matplotlib=1.5 pandas=0.18 scikit-learn=0.18 pytest pytest-cov pytest-mock pytest-timeout
Collecting package metadata (repodata.json): done
Solving environment: / 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                                                                                              

UnsatisfiableError: The following specifications were found to be incompatible with each other:
```
Maybe we should bump our minimum requirements to whatever the minimums are we can get with `conda`...?

massich, 2019-09-04:
> Maybe we should bump our minimum requirements to whatever the minimums are we can get with conda...?

That sounds good to me !!! :D

Maybe since we can be aggressive in bumping the dependencies since we'll be releaseing 'soonish'.

PR #6735 (2019-09-05): MRG, MAINT: Bump minimum versions
It turns out that our "old deps" job was not actually testing old dependencies. This should fix it.

It also bumps our minimums for NumPy, SciPy, Pandas, and matplotlib by one minimum version (but for matplotlib this turns out to require a major version bump 1.5->2.0). This seems like the most sustainable path since it does not really seem possible to CI-test our currently listed minimum dependencies, which is a problem.

If this is okay, I'll then go through `fixes` (and maybe `viz/utils.py`?) and axe anything we no longer need.

- [x] Bump versions
- [x] Che [... cut]
closed 2019-09-05 by larsoner