mne-tools/mne-python #6634: remove gray background from ICA.plot_sources

drammock, 2019-08-06:
I don't mind the gray background for things like `ICA.plot_properties()` where several subplots of different types are all generated in one figure --- in tutorials and notebooks I think it makes it clearer that the several figures were generated by one function/method call.  But `ICA.plot_sources()` generates a single graph (similar to `Raw.plot()`) so I don't see why it's background should be gray.

cc @cbrnr

agramfort, 2019-08-06:
Ok for me

drammock, 2019-08-06:
actually I think this is a bug (possibly a sphinx-gallery bug?).  `ICA.plot_sources()` doesn't *always* have a gray background; seems like maybe it only happens when there's a previous call to `ICA.plot_properties()` in the same notebook cell...  but I can't replicate it even when downloading the `.ipynb` file from https://14567-1301584-gh.circle-artifacts.com/0/dev/auto_tutorials/preprocessing/plot_40_artifact_correction_ica.html and running it locally.

@larsoner @choldgraf do you have any insight into whether this might be a sphinx-gallery issue? Or something weird about CircleCI?  As far as I can tell we don't change `first_notebook_cell` anywhere, so I'm kinda stumped what is causing this.

larsoner, 2019-08-06:
Can you replicate locally with `PATTERN=... make html_dev-pattern`? If so then we have a way forward for debugging.

Before each example is executed, a set of `reset_modules` actions is called:

https://sphinx-gallery.github.io/configuration.html#reset-modules

https://github.com/sphinx-gallery/sphinx-gallery/blob/master/sphinx_gallery/scrapers.py#L306-L309

If you put a `plt.rcdefaults()` in between the two calls, does it fix `make html_dev-pattern`? If so, then perhaps there is some `rcParams` thing being inadvertently messed with in `plot_sources` or somewhere else.

drammock, 2019-08-06:
yes, can replicate with `PATTERN=... make html_dev-pattern`.  CircleCI:

[image]

Local:

[image]

Local with `rcdefaults()` between `plot_properties()` and `plot_sources()`:

[image]

larsoner, 2019-08-06:
I realized that it was probably related to this:

https://github.com/sphinx-gallery/sphinx-gallery/blob/f270049dbc209f1d3bd5598c9fac19b5f665ecee/sphinx_gallery/scrapers.py#L89-L95

See https://github.com/sphinx-gallery/sphinx-gallery/pull/527

drammock, 2019-08-06:
this is not an MNE-Python bug; will be fixed when sphinx-gallery/sphinx-gallery#527 is merged in.

closed 2019-08-06 by drammock