mne-tools/mne-python #6644: RFC: in search of the perfect sample data

drammock, 2019-08-08:
After revising lots of tutorials, I've formed some opinions about what the ideal sample dataset would look like. Here are some thoughts; please feel free to edit / add to the list; maybe if we reach some consensus we can find a dataset that meets the criteria and could replace `sample` dataset in most/all of the tutorials.

- probably should be neuromag and include fine cal, crosstalk, and a cHPI `.pos` file (current sample data lacks `.pos`)
- has mag, grad, EEG, EOG, and ECG (current sample data lacks `ECG`)
- has multiple subjects (useful for demoing `corrmap`, stats, ...)
- multiple runs per subject maybe? (for things like concatenating raws)
- a single Raw recording is ~90 seconds max, and/or has a low-ish acquisition sampling frequency (so we're not constantly cropping to save memory)
- raw file does not already contain projectors; separate empty-room file included
- I think the task (L/R aud/vis + catch trials) is pretty good as a demo, but might be nice to have reaction time data included (for things like the epochs image overlay). I'm certainly open to other ideas about the paradigm though.

larsoner, 2019-08-08:
> - probably should be neuromag ...
> ...
> - raw file does not already contain projectors

These two are going to be almost mutually exclusive in standard Neuromag acquisition setups, and thus to satisify this requirement, be an outlier. Every Neuromag file I've seen has projectors (even ones from MGH) because there are empty room projectors added during acquisition. You can probably tweak the acquisition not to add these, but then our dataset will be unlike ones people will be used to getting.

`raw.del_proj()` where we need it seems like the reasonable thing to do instead.

> I think the task (L/R aud/vis + catch trials) is pretty good as a demo, but might be nice to have reaction time data included (for things like the epochs image overlay).

IIRC there were some catch trials in `sample`, so you could perhaps already have some basic RTs there. I don't think anyone has looked.

> a single Raw recording is ~90 seconds max, and/or has a low-ish acquisition sampling frequency (so we're not constantly cropping to save memory)

I think the thing to do is probably to use a low acq sample rate. The ERPs after 90 sec if you want multiple conditions will probably not look too good.

I do have cHPI + Neuromag MEG + EEG + ECG + EOG + auditory + visual + catch trials + empty-room for multiple subjects (functional localizer basically) that were recorded a long time ago at UW. It's possible we'd actually have a dozen or so, I'd have to check, but I know we have at least two. We could anonymize these, downsample them to 100 Hz (even though this loses some event precision), and even remove the acquisition projectors if you want (though see above about why it's not a great idea).

There aren't multiple runs per subject, but there is no reason we couldn't fake it by cropping it so that there were two "runs" and saving these to FIF.

drammock, 2019-08-08:
I guess this reveals my ignorance about empty room recordings... I thought that sometimes they were done *after* the subject left, so I don't understand how the projectors got into the experimental raw file *during acquisition*.  (incidentally I'm also confused why the sample data's `erm` file itself already contains projectors...)

larsoner, 2019-08-08:
During annual maintenance (or sometime) an Elekta tech will make an empty room recording, compute projectors, and set these as acquisition projectors that will be used (until the next time they are updated, which might be never, might be in a year, etc.). The goal is to make it so that huge deflections due to env noise do not make it impossible to see real activity during acquisition of subject data.

Empty-room projectors computed from empty room recordings the same day are better because they give more accurate projectors.

jasmainak, 2019-08-08:
what is the reason for this move? multiple runs, multiple subjects starts to move towards group analysis / BIDS style analysis. The other conditions are more or less satisfied by the existing sample dataset?

You can find many datsets here with multiple subjects and runs: https://openneuro.org/public/datasets

larsoner, 2019-08-08:
> The other conditions are more or less satisfied by the existing sample dataset?

On top of the single-subject-ness that you mention, from the above it (at least) has no ECG, has no cHPI, and you need to crop to fit into memory to process it. We could "fix the" single-run-ness via cropping but it's nicer for backward compat not to mess with it.

larsoner, 2019-08-08:
The other consideration -- which is implicit but we might as well make explicit -- is that we want a dataset that is *small* but has all of these features. OpenNeuro datasets are likely to be either single subject or complete datasets from real paradigms that people did "for science" -- meaning that they are likely going to be much larger than they would need to be if we made/got one that was done "for demoing as much functionality as possible with minimal size".

BTW I didn't realize we had published with the dataset I'm discussing, but it's the one listed under "task" here:

https://www.frontiersin.org/articles/10.3389/fnins.2014.00330/full

It seems pretty close to what we want.

agramfort, 2019-08-09:
if we switch to a new dataset I would conisder a bids formatted dataset.
Now moving away from sample will force us to change a lot of material in
mne...
if we switch to a bids formatted sample we would fake 2 runs by cropping
the file...

>

drammock, 2019-08-09:
I agree it will force a lot of changes @agramfort. I'm thinking this as a long-term goal not immediate.  But it's a way of simplifying a lot of tutorials by eliminating repetitive code that is unrelated to each tutorial's purpose, and avoiding digressions like "now we'll fake having multiple subjects by pretending that different parts of the same recording are actually different subjects" (which is always distracting, and sometimes less convincing like when you're trying to show that artifact template matching with `corrmap` works across subjects). In the case of `corrmap` I ended up introducing a whole new dataset (EEGBCI) rather than spoofing multiple subjects, but that also adds an extra paragraph of explaining what the dataset is and why we're switching to it --- again, a distraction from the main point of the lesson (which is "using ICA for artifact repair", not "what is the EEGBCI dataset like").

agramfort, 2019-08-10:
I understand. Things I like about sample and that should be present
with a new dataset
are: clear bad channels (noisy and flat), outlier in dig point.

jona-sassenhagen, 2019-08-24:
A continuous variable (like RT) would allow demoing a lot of things much easier indeed.