mne-tools/mne-python #6645: mne-python is either GPL or non-free now due to mne/io/edf/gdf_encodes.txt

yarikoptic, 2019-08-08:
mne/io/edf/gdf_encodes.txt from biosig says
```
# BioSig is free software; you can redistribute it and/or
# modify it under the terms of the GNU General Public
# License as published by the Free Software Foundation; either
# Version 3 of the License, or (at your option) any later version.
#
# Alternative license: 
# This table of event codes can be used in any application 
# (without the restrictions of GPL) under the condition that this table 
# is not modified, changed, converted or transformed. No 
# derivative of any kind are allowed. Suggestions for improvement 
# and changes should be addressed to the author. 
```
so either then mne-python is effectively under GPL license or non-free (no modifications allowed)... what it is ? ;)  May be @schloegl could relax this restriction (I am really not sure why not to allow changes/fixes etc)?

cbrnr, 2019-08-09:
AFAIK we didn't change the table, so we use the alternative license. Unlike the GPL, this alternative license only applies to this single file and does not automatically transfer to MNE.

schloegl, 2019-08-09:
On 08.08.19 22:01, Yaroslav Halchenko wrote:
> mne/io/edf/gdf_encodes.txt from biosig says
> ```
> # BioSig is free software; you can redistribute it and/or
> # modify it under the terms of the GNU General Public
> # License as published by the Free [...]

I do allow changes - under the terms of GPLv3.

The code table is used to standardize codes for events, markers and
annotations. Compatibility can only be maintained if the table is
normative.  Therefore, this event table must stay public and nobody
should use a modified, non-public version of this table.  The BSD
license (used by MNE) is not sufficient to guarantee this.

Alois

agramfort, 2019-08-09:
I understand but how do you allow then the existence of BSD / MIT licensed
readers for GDF files?

yarikoptic, 2019-08-09:
> Unlike the GPL, this alternative license only applies to this single file and does not automatically transfer to MNE

" is not modified, changed, converted or transformed" restrictions make it non-free.  Moreover "converted" is IMHO a bad term, since under some stretch of interpretation then it is not usable at all since use of those would typically require internal (to the codebase) conversion to other data types/representations.  IMHO trying to enforce it this way provides only disadvantages/inconveniences without actually solving any problem (I really do not see people modifying this file for their own good). tables of values for formats are known to exist in other cases (e.g. iirc

cbrnr, 2019-08-09:
I don't understand the reasoning either, but if we can't use this file then I suggest we remove it and return only the event codes instead of the text mappings from the file.

agramfort, 2019-08-10:
@schloegl are you asking us to remove the file with the consequence that mne cannot properly read GDF files?

closed 2019-08-13 by larsoner

agramfort, 2019-08-13:
Maybe we should backport so yarik can cut a Debian release of 0.18 ? Or we wait for 0.19 release in September

larsoner, 2019-08-13:
0.19 is only one month away, my vote is just to wait

PR #6655 (2019-08-13): [MRG] Remove GDF event table
Fixes #6645. This removes the GDF event table file because of restrictive licensing (either GPL3 or non-free, none of which can be used with MNE). This has the effect that we can still read GDF annotations, but we don't provide a text description anymore. Instead, we can only provide event IDs, which should be sufficient for most purposes. If a user must have the text descriptions, we could add some kind of documentation how to download the original file and quickly convert event IDs to text descriptions.