mne-tools/mne-python #6647: BUG, VIZ: can't use `fig.subplots_adjust` on `raw.plot` figure

drammock, 2019-08-08:
well, you can, but it looks awful:

[image]

Note especially the `Help` button and the extra-wide channel and time scrollbars. This resulted from:
```
eeg_picks = mne.pick_types(raw.info, meg=False, eeg=True)
fig = raw.plot(proj=False, order=eeg_picks, n_channels=len(eeg_picks))
fig.subplots_adjust(top=0.94)
fig.suptitle('No reference', size='xx-large', weight='bold')
```
Possible solutions include:

- leaving enough room for a title at the top edge by default (and possibly putting the title on the plot instead of / in addition to in the window titlebar)
- debugging to figure out why it gets so messed up currently, and fixing it so that `subplots_adjust` works more or less normally

cbrnr, 2019-08-09:
I'd rather not add additional space for a title within the figure, because in interactive use this just takes away precious screen estate that should be used for the signal traces. So I'll definitely be +1 for debugging/improving the figure structure without changing the current layout. Alternatively, we might add options to hide the buttons (Help and Proj) as well as the scroll bars to produce a much simpler figure with basically only one axis containing the signals. This will make it straightforward to manipulate for further use.

larsoner, 2019-08-09:
> Alternatively, we might add options to hide the buttons (Help and Proj) as well as the scroll bars to produce a much simpler figure with basically only one axis containing the signals. 

+1 for "z" meaning "zen" mode :)

But yes I agree we should fix how we lay out the figure somehow so that `subplots_adjust` works

drammock, 2019-08-13:
I dug into the code today, and it looks like this might be possible.  My current understanding of how this works is that:

1. main axes, scrollbars, and help button get initialized as subplots in `_prepare_mne_browse_raw`
2. positions of everything get modified in `_layout_figure()` any time the window is resized

I *think* we could get `subplots_adjust` to work correctly if:
1.  `subplots_adjust` fires an event that we can listen for (not yet sure if it does), and 
2. `_layout_figure` is called after that event, and
3. `_layout_figure` is modified to take `fig.subplotpars` into account

I'll take a little time tomorrow to follow up on those points.

drammock, 2019-08-14:
I've made some progress here. on [my branch](https://github.com/drammock/mne-python/blob/subplots-adjust/mne/viz/utils.py#L525-L643) I split `_layout_figure` into two functions (the new one called `_update_layout`. 

There's a lot of duplicated code there that needs refactoring, but for now here's what works:
```
fig = raw.plot()
fig.subplots_adjust(top=0.9)
```
The plot will still be wonky, but after resizing the window it will look right, and the wider top border will persist through subsequent resizes.  I think I can fix the initial wonkyness by taking what I do [here](https://github.com/drammock/mne-python/blob/subplots-adjust/mne/viz/utils.py#L638-L642) and doing something similar for each `axes`'s position attributes instead of just for the `subplotpars`.

drammock, 2019-08-15:
May have hit a wall on this one, so I've resorted to stackoverflow: https://stackoverflow.com/q/57503564/1664024

PR #6670 (2019-08-27): MRG, ENH: make subplots_adjust work with {raw,epochs}.plot() and ica.plot_sources()
closes #6647 

- [x] raw
- [x] epochs
- [x] ICA
closed 2019-08-27 by cbrnr