mne-tools/mne-python #6652: `tmax` optionally exclusive in Epochs and cropping

milan-rybar, 2019-08-12:
Currently, `tmax` parameter in `Epochs` and `mne.epochs.BaseEpochs.crop` is inclusive. It would be convenient to be rather exclusive in some cases.
For example, in the example for "Decoding in time-frequency space data using the Common Spatial Pattern (CSP)" (https://mne-tools.github.io/stable/auto_examples/decoding/plot_decoding_csp_timefreq.html?highlight=csp), the grid search is used in the time domain by `crop` method:
```python
        # Crop data into time-window of interest
        X = epochs.copy().crop(w_tmin, w_tmax).get_data()
```
But this results in reusing one sample between two following windows!

What do you think? Is there any reason why it is strictly inclusive?

larsoner, 2019-08-12:
> Is there any reason why it is strictly inclusive?

It has been that way for many years so we should not change the default behavior (it could break a lot of people's code). It is however a pain to do `epochs.crop(start, stop - 1. / epochs.info['sfreq'])` so I wouldn't mind adding a `include_last=True` (by default) keyword argument.

milan-rybar, 2019-08-13:
I am for adding a keyword argument such as `tmax_inclusive` or `include_last` with `True` as default value.

agramfort, 2019-08-13:
Ok for a include_tmax param in crop method

PR #6683 (2019-08-20): MRG, ENH: Add include_tmax to cropping methods
Closes #6652.

I stopped adding tests after doing Raw/Epochs/Evoked because they should be general enough to catch the others (SourceEstimate, AverageTFR, Dipole).
closed 2019-08-20 by larsoner