mne-tools/mne-python #6654: code understanding

as485, 2019-08-13:
Hi there, When I was learning and trying to understand the function "test_savgol_filter()" in mne-python/mne/tests/test_evoked.py, I was confused about the description of the its delivered function. My understanding of this code snippet is "test evoked data as title". Do I understand that right?

mmagnuski, 2019-08-13:
`test_savgol_filter()` is part of the test suite which ensures that new changes to mne don't break existing functionality. `test_savgol_filter()`, as the name suggests, tests the validity of savgol filtering for `Evoked` objects (the operation you can invoke by doing `evoked.savgol_filter(h_freq)`). You can read more about savgol filter [here](https://en.wikipedia.org/wiki/Savitzky%E2%80%93Golay_filter).

as485, 2019-08-14:
Thanks for your response very much. I will read the code more carefully.

closed 2019-08-14 by mmagnuski