mne-tools/mne-python #6656: Epochs rejecting good data?

Sarah335, 2019-08-13:
Hi all,
I'm doing some connectivity analysis on resting state MEG data, and am attempting to epoch the pre-processed files into "fake" events 10s in length. The mne.Epochs method seems to be dropping "bad" epochs, even though there appears to be sufficient data present (although it's not clear to me how insufficient data is defined). Attached is the relevant snippet of my code and the output of drop_log and get_data. Any insight would be appreciated.

larsoner, 2019-08-13:
Your `events` are probably occuring before the start of the raw data. `events[:, 0]` are relative to `raw.first_samp`. Does everything work properly when you use:
```
sample = raw.first_samp  # instead of sample = 5. * sfreq
```

BTW it looks like you are recreating the functionality of [mne.make_fixed_length_events](https://martinos.org/mne/stable/generated/mne.make_fixed_length_events.html) (which already takes `raw.first_samp` into account) with your code, so it would be better just to use that function.

Sarah335, 2019-08-13:
Thanks for your quick response.
mne.make_fixed_length_events fixed more problems than I was expecting! I'll adjust a few codes this way :)

closed 2019-08-13 by Sarah335