mne-tools/mne-python #6657: read_* functions should support pathlib objects

hoechenberger, 2019-08-13:
#### Describe the problem
This currently throws an error:
```python

import pathlib
import mne

data_dir = pathlib.Path('/Users/hoechenberger/Projects/TaLa/'
                        'Data/EEG/Preprocessed')

data_file = data_dir / 'TaLa_460_preped-epo.fif'
data = mne.read_epochs(data_file)
```

Generates:
```python
Reading File-like PosixPath('/Users/hoechenberger/Projects/TaLa/Data/EEG/Preprocessed/TaLa_460_preped-epo.fif') ...
Traceback (most recent call last):
  File "/Users/hoechenberger/Library/Preferences/PyCharm2019.2/scratches/scratch_45.py", line 8, in <module>
    data = mne.read_epochs(data_file)
  File "</Users/hoechenberger/Development/mne-python/mne/externals/decorator.py:decorator-gen-179>", line 2, in read_epochs
  File "/Users/hoechenberger/Development/mne-python/mne/utils/_logging.py", line 89, in wrapper
    return function(*args, **kwargs)
  File "/Users/hoechenberger/Development/mne-python/mne/epochs.py", line 2347, in read_epochs
    return EpochsFIF(fname, proj, preload, verbose)
  File "</Users/hoechenberger/Development/mne-python/mne/externals/decorator.py:decorator-gen-180>", line 2, in __init__
  File "/Users/hoechenberger/Development/mne-python/mne/utils/_logging.py", line 89, in wrapper
    return function(*args, **kwargs)
  File "/Users/hoechenberger/Development/mne-python/mne/epochs.py", line 2413, in __init__
    fid, tree, _ = fiff_open(fname, preload=preload)
  File "</Users/hoechenberger/Development/mne-python/mne/externals/decorator.py:decorator-gen-6>", line 2, in fiff_open
  File "/Users/hoechenberger/Development/mne-python/mne/utils/_logging.py", line 89, in wrapper
    return function(*args, **kwargs)
  File "/Users/hoechenberger/Development/mne-python/mne/io/open.py", line 119, in fiff_open
    fid = _fiff_get_fid(fname)
  File "/Users/hoechenberger/Development/mne-python/mne/io/open.py", line 50, in _fiff_get_fid
    fid.seek(0)
  File "/Users/hoechenberger/Development/mne-python/mne/io/open.py", line 33, in seek
    return self.fid.seek(offset, whence)
AttributeError: 'PosixPath' object has no attribute 'seek'
```

#### Describe your solution
My current workaround: convert the `PosixPath` object to a string:
```python
data = mne.read_epochs(str(data_file))
```

#### Describe possible alternatives
Ideally, there would be "built-in" `pathlib` support in MNE.

larsoner, 2019-08-13:
This is a relatively new problem because we basically used to assume that everything was a str/path-like. Now we also support file-like objects.

Maybe the most backward compatible thing to do would be to check if the object is file like with:
```
isinstance(something, (TextIOBase, BufferedIOBase, RawIOBase, IOBase))
```
and if not, cast with `str` and treat as a path.

@hoechenberger if this sounds reasonable to you, do you want to try making this change?

hoechenberger, 2019-08-13:
@larsoner I just checked how this was solved in PsychoPy, and we're using a small converter function there that converts pathlib objects to strings, but only works withPython 3.6+:
https://github.com/psychopy/psychopy/blob/86046113feaf8afbbeb65aec0cc9a0bc0acde580/psychopy/tools/filetools.py#L283

Could that be a valid approach for MNE too?

larsoner, 2019-08-13:
In your approach, we would treat/cast `str` and `PosixPath` as `str`, and everything else `file-like`, which seems less general. For just one example, `tmpdir` pytest fixtures can be `str`-ified properly, but would need to be added as yet another case in your approach. Having "treat as str" as the default use case / mode should in this sense be more backward compatible, yet still work with any file-like object (with proper inheritance patterns at least).

If the file-like object approach ends up not working for some file-like objects, we can change it (e.g., by checking for `.read()` and `.seek()` methods) and this seems easier than trying to keep up with all `str`-like types we should support, or supported implicitly in the past.

hoechenberger, 2019-08-16:
Sounds reasonable. Will try to look into this, but cannot make any promises on a time frame. However, this issue seems to be sufficiently annoying to drive me to actually fix it ;)

PR #6682 (2019-08-20): MRG, ENH: Better support for str-like objects
Closes #6657. cc @hoechenberger
closed 2019-08-20 by agramfort