mne-tools/mne-python #6666: Support for Nexus amplifiers (MindMedia)

tamalotes, 2019-08-15:
#### Describe the problem
The Nexus amplifier formats are not supported.

#### Describe your solution
Support Nexus supported file formats which are: *.mbd and *.bcd.

#### Describe possible alternatives
A format translator from Nexus format to a supported format.

#### Additional context
I am not familiar with the Nexus file formats. I will continue to search if there is any documentation or plugin to these formats that helps with the support and possible implementation.

drammock, 2019-08-15:
based on [this webpage](https://mindmedia.freshdesk.com/support/solutions/articles/36000077097-nexus-and-biotrace-interfacing-options) and [this PDF](http://81.169.139.131/download/documentation/UserManual_MM_BioTrace-V2018A_EN.pdf) (page 62 and following) it looks like the MindMedia BioTrace software can export EDF format.  If you have access to BioTrace that would be the quickest solution since MNE-Python can already read EDF.  Otherwise you may need to contact MindMedia to get a description of the `mbd` and `bcd` file formats, since they don't seem to be described on their website or in the device or software manuals.

tamalotes, 2019-08-16:
I  will try this over the weekend and report back. Thanks