mne-tools/mne-python #6667: Error with mne.bem.make_watershed_bem using the development version of Freesurfer

cedricsimar, 2019-08-16:
In the process of creating the boundary-element model meshes, I run into the following error when executing `mne.bem.make_watershed_bem('fsaverage', overwrite=True)` with the development version of Freesurfer (note that the error does not appear in the stable version):

`File "/home/rbcx/miniconda3/envs/eeg/lib/python3.6/site-packages/nibabel/freesurfer/io.py", line 179, in read_geometry coords = np.fromfile(fobj, ">f4", vnum * 3).reshape(vnum, 3) ValueError: cannot reshape array of size 92098 into shape (650288644,3)`

The complete console output is the following:

```
Running mri_watershed for BEM segmentation with the following parameters:
 
SUBJECTS_DIR = /usr/local/freesurfer/subjects
SUBJECT = fsaverage
Results dir = /usr/local/freesurfer/subjects/fsaverage/bem/watershed
 
Running subprocess: mri_watershed -useSRAS -surf /usr/local/freesurfer/subjects/fsaverage/bem/watershed/fsaverage /usr/local/freesurfer/subjects/fsaverage/mri/T1.mgz /usr/local/freesurfer/subjects/fsaverage/bem/watershed/ws
 
Mode:          use surfaceRAS to save surface vertex positions
Mode:          Saving out BEM surfaces
 
*********************************************************
The input file is /usr/local/freesurfer/subjects/fsaverage/mri/T1.mgz
The output file is /usr/local/freesurfer/subjects/fsaverage/bem/watershed/ws
 
*************************WATERSHED**************************
Sorting...
      first estimation of the COG coord: x=127 y=142 z=110 r=85
      first estimation of the main basin volume: 2635318 voxels
      global maximum in x=96, y=97, z=106, Imax=255
      CSF=16, WM_intensity=110, WM_VARIANCE=0
      WM_MIN=109, WM_HALF_MIN=109, WM_HALF_MAX=110, WM_MAX=110
      preflooding height equal to 25 percent
done.
Analyze...
 
      main basin size= 2220668 voxels, voxel volume =1.000
                     = 2220668 mmm3 = 2220.668 cm3
done.
PostAnalyze...
      ***** 0 basin(s) merged in 1 iteration(s)
      ***** 0 voxel(s) added to the main basin
done.
 
****************TEMPLATE DEFORMATION****************
 
      second estimation of the COG coord: x=129,y=126, z=109, r=12780 iterations
^^^^^^^^ couldn't find WM with original limits - expanding ^^^^^^
 
   GLOBAL      CSF_MIN=19, CSF_intensity=38, CSF_MAX=51 , nb = 38178
 (2) Problem with the least square interpolation in GM_MIN calculation.
   
                     CSF_MAX  TRANSITION  GM_MIN  GM
    GLOBAL    
  before analyzing :    51,      44,        21,   67
  after  analyzing :    42,      44,        44,   49
      mri_strip_skull: done peeling brain
      highly tesselated surface with 10242 vertices
      matching...78 iterations
 
*********************VALIDATION*********************
curvature mean = -0.012, std = 0.010
curvature mean = 75.679, std = 7.749
 
No Rigid alignment: -atlas Mode Off (basic atlas / no registration)
      before rotation: sse = 5.32, sigma = 9.30
      after  rotation: sse = 5.32, sigma = 9.30
Localization of inacurate regions: Erosion-Dilation steps
      the sse mean is  6.22, its var is 11.52  
      before Erosion-Dilatation  3.79% of inacurate vertices
      after  Erosion-Dilatation  0.00% of inacurate vertices
      Validation of the shape of the surface done.
Scaling of atlas fields onto current surface fields
 
********FINAL ITERATIVE TEMPLATE DEFORMATION********
Compute Local values csf/gray
Fine Segmentation...37 iterations
 
      mri_strip_skull: done peeling brain
 
Brain Size = 2029869 voxels, voxel volume = 1.000 mm3
           = 2029869 mmm3 = 2029.869 cm3
 
      outer skin surface matching...173 iterations
 
******************************
Saving /usr/local/freesurfer/subjects/fsaverage/bem/watershed/ws
done
mri_watershed utimesec    288.051490
mri_watershed stimesec    0.183948
mri_watershed ru_maxrss   472268
mri_watershed ru_ixrss    0
mri_watershed ru_idrss    0
mri_watershed ru_isrss    0
mri_watershed ru_minflt   91446
mri_watershed ru_majflt   0
mri_watershed ru_nswap    0
mri_watershed ru_inblock  0
mri_watershed ru_oublock  35688
mri_watershed ru_msgsnd   0
mri_watershed ru_msgrcv   0
mri_watershed ru_nsignals 0
mri_watershed ru_nvcsw    5558
mri_watershed ru_nivcsw   3861
mri_watershed done
Traceback (most recent call last):
  File "mne_processing.py", line 52, in <module>
    mne.bem.make_watershed_bem('fsaverage', overwrite=True)
  File "</home/rbcx/miniconda3/envs/eeg/lib/python3.6/site-packages/mne/externals/decorator.py:decorator-gen-45>", line 2, in make_watershed_bem
  File "/home/rbcx/miniconda3/envs/eeg/lib/python3.6/site-packages/mne/utils/_logging.py", line 89, in wrapper
    return function(*args, **kwargs)
  File "/home/rbcx/miniconda3/envs/eeg/lib/python3.6/site-packages/mne/bem.py", line 1128, in make_watershed_bem
    read_metadata=True)
  File "</home/rbcx/miniconda3/envs/eeg/lib/python3.6/site-packages/mne/externals/decorator.py:decorator-gen-36>", line 2, in read_surface
  File "/home/rbcx/miniconda3/envs/eeg/lib/python3.6/site-packages/mne/utils/_logging.py", line 89, in wrapper
    return function(*args, **kwargs)
  File "/home/rbcx/miniconda3/envs/eeg/lib/python3.6/site-packages/mne/surface.py", line 580, in read_surface
    ret = _get_read_geometry()(fname, read_metadata=read_metadata)
  File "/home/rbcx/miniconda3/envs/eeg/lib/python3.6/site-packages/nibabel/freesurfer/io.py", line 179, in read_geometry
    coords = np.fromfile(fobj, ">f4", vnum * 3).reshape(vnum, 3)
ValueError: cannot reshape array of size 92098 into shape (650288644,3)
```

For now, the only solution I found was to use the stable version (6.0.0) of Freesurfer.

agramfort, 2019-08-16:
apparently it's the surface file produced whose format has changed and
causes nibabel reader to crash. Can you share the file?
It's more likely to be a nibabel issue than an MNE issue.

>

larsoner, 2019-08-17:
I don't think this will fix any development version surface reading

larsoner, 2019-08-17:
I mean I don't think my PR did

larsoner, 2019-08-20:
This is a FreeSurfer bug, see https://github.com/freesurfer/freesurfer/issues/699

closed 2019-08-20 by larsoner

cedricsimar, 2019-08-21:
Thanks for the info @larsoner

PR #6668 (2019-08-17): MRG, ENH: Add options to mne watershed_bem
closes #6667

This PR for `mne watershed_bem`:

1. Adds control over the `-T1` option (which was otherwise always tied to our `--gcaatlas` option)
2. Adds control over where the brainmasked output ends up.
3. Allows passing `-v orig.mgz` instead of just `-v orig` (be more lenient)
4. Works with newer Freesurfer (the `gcaatlas` option assumed that a file from an older FS version was there, this finds whichever is newest even though there will probably only ever be one)

This is because for some subjects I've wanted to do:
```
mri_watershed -useSRAS -br [... cut]