mne-tools/mne-python #6675: stats.spatio_temporal_cluster_1samp_test() fails with an index error

hoechenberger, 2019-08-19:
#### Describe the bug
`stats.spatio_temporal_cluster_1samp_test()` fails with an index error. Unfortunately I didn't manage to put together a MWE yet, but hopefully the provided information is sufficient to track things down?

#### Steps and/or code to reproduce
```python
src = mne.read_source_spaces(surface_source_space_file)
connectivity = mne.spatial_src_connectivity(src)

# Assemble STCs from individual subjects.
# `stcs` is a list of SourceEstimates.
# The resulting shape of the ndarray is (n_subj, n_timepoints, n_vertices)
stcs_for_stats = [s.data.T for s in stcs]
stcs_for_stats = np.array(stcs_for_stats)

mne.stats.spatio_temporal_cluster_1samp_test(
    X=stcs_for_stats,
    connectivity=connectivity,
    n_jobs=1,
    verbose=True)
```

#### Expected results
The function should return the results of the test.
#### Actual results
```python
Using a threshold of 2.109816
stat_fun(H1): min=2.351713 max=20.239683
/Users/hoechenberger/Projects/TaLa/Code/Analysis/EEG/source_70_make_inverse_solution.py:132: RuntimeWarning: Provided stat_fun does not treat variables independently. Setting buffer_size to None.
  verbose=True)
Running initial clustering
Traceback (most recent call last):
  File "/Users/hoechenberger/miniconda3/envs/TaLa-source-loc/lib/python3.7/site-packages/IPython/core/interactiveshell.py", line 3326, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-2-7331a14a6e31>", line 1, in <module>
    runfile('/Users/hoechenberger/Projects/TaLa/Code/Analysis/EEG/source_70_make_inverse_solution.py', wdir='/Users/hoechenberger/Projects/TaLa/Code/Analysis/EEG')
  File "/Applications/PyCharm.app/Contents/helpers/pydev/_pydev_bundle/pydev_umd.py", line 197, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "/Applications/PyCharm.app/Contents/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/Users/hoechenberger/Projects/TaLa/Code/Analysis/EEG/source_70_make_inverse_solution.py", line 132, in <module>
    verbose=True)
  File "</Users/hoechenberger/miniconda3/envs/TaLa-source-loc/lib/python3.7/site-packages/mne/externals/decorator.py:decorator-gen-218>", line 2, in spatio_temporal_cluster_1samp_test
  File "/Users/hoechenberger/miniconda3/envs/TaLa-source-loc/lib/python3.7/site-packages/mne/utils/_logging.py", line 88, in wrapper
    return function(*args, **kwargs)
  File "/Users/hoechenberger/miniconda3/envs/TaLa-source-loc/lib/python3.7/site-packages/mne/stats/cluster_level.py", line 1382, in spatio_temporal_cluster_1samp_test
    check_disjoint=check_disjoint, buffer_size=buffer_size)
  File "</Users/hoechenberger/miniconda3/envs/TaLa-source-loc/lib/python3.7/site-packages/mne/externals/decorator.py:decorator-gen-217>", line 2, in permutation_cluster_1samp_test
  File "/Users/hoechenberger/miniconda3/envs/TaLa-source-loc/lib/python3.7/site-packages/mne/utils/_logging.py", line 89, in wrapper
    return function(*args, **kwargs)
  File "/Users/hoechenberger/miniconda3/envs/TaLa-source-loc/lib/python3.7/site-packages/mne/stats/cluster_level.py", line 1264, in permutation_cluster_1samp_test
    buffer_size=buffer_size)
  File "/Users/hoechenberger/miniconda3/envs/TaLa-source-loc/lib/python3.7/site-packages/mne/stats/cluster_level.py", line 875, in _permutation_cluster_test
    show_info=True)
  File "/Users/hoechenberger/miniconda3/envs/TaLa-source-loc/lib/python3.7/site-packages/mne/stats/cluster_level.py", line 431, in _find_clusters
    ndimage)
  File "/Users/hoechenberger/miniconda3/envs/TaLa-source-loc/lib/python3.7/site-packages/mne/stats/cluster_level.py", line 475, in _find_clusters_1dir_parts
    t_power, ndimage)
  File "/Users/hoechenberger/miniconda3/envs/TaLa-source-loc/lib/python3.7/site-packages/mne/stats/cluster_level.py", line 527, in _find_clusters_1dir
    clusters = _get_clusters_st(x_in, connectivity, max_step)
  File "/Users/hoechenberger/miniconda3/envs/TaLa-source-loc/lib/python3.7/site-packages/mne/stats/cluster_level.py", line 254, in _get_clusters_st
    row, col = np.unravel_index(cl_goods, (n_times, n_src))
ValueError: index 4917600 is out of bounds for array with size 4917600
```
#### Additional information
```python
In[3]: stcs_for_stats.shape
Out[3]: (18, 601, 8195)
In[11]: connectivity.shape
Out[11]: (8196, 8196)
```

Inside `mne.stats.cluster_level._get_clusters_st()`:
```python
In[3]: len(x_in)
Out[3]: 4925195
In[4]: n_src
Out[4]: 8196
In[5]: n_times
Out[5]: 600
In[7]: len(cl_goods)
Out[7]: 4925195
```

larsoner, 2019-08-19:
Can you share `X` and `connectivity`?

hoechenberger, 2019-08-19:
@larsoner Since this is actual, unpublished data, may I share this privately with you?

larsoner, 2019-08-19:
Sure, feel free to email a link

hoechenberger, 2019-08-19:
Done. Thanks for your help, @larsoner!

hoechenberger, 2019-08-22:
@larsoner:
I did like you proposed in the email and used the source space after forward computation
```python
src = fwd['src']
```
instead of the original source space. With this, it works! 

Turns out that during creation of the forward model, one data point was being dropped:
`1 source space point omitted because of the    5.0-mm distance limit.`
so I ended up having a discrepancy. Wow, what a stupid mistake, yet it's so easy to make! Good thing this one point had been omitted, otherwise I would have never found out!

---

Now, when running `mne.spatial_src_connectivity()` on the _correct_ source space, I get this warning:
`RuntimeWarning: 0.0% of original source space vertices have been omitted, tri-based connectivity will have holes.
Consider using distance-based connectivity or morphing data to all source space vertices.`

The original source spacing was `oct6`, and `make_bem_model()` was called with `ico=4`. Is this warning save to ignore? With matching source and BEM spacings (e.g., `ico4` and `ico=4`, respectively),  the warning disappears.

closed 2019-08-22 by larsoner

larsoner, 2019-08-22:
> I get this warning:

Yes this is your one vertex creating a hole. You can use dist-based connectivity instead to help mitigate the problems the hole creates with mesh-edge-based connectivity. If you created your original source space point with `add_dist=True` you can set e.g. `dist=0.007` in `spatial_src_connectivity`.

> With matching source and BEM spacings (e.g., ico4 and ico=4, respectively), the warning disappears.

This is probably because using an ico-4 spacing you've removed a ton of source space points, one of which was the one that got excluded during forward computation. I would not recommend this.

hoechenberger, 2019-08-25:
@larsoner 
> If you created your original source space point with `add_dist=True` you can set e.g. `dist=0.007` in `spatial_src_connectivity`.

Thank you, this actually makes the warning go away! :)

PR #6680 (2019-08-22): MRG, DOC: Better error message for connectivity problems
Closes #6675. cc @hoechenberger 

Along the way also modernizes some existing raises checks.