mne-tools/mne-python #6676: raw.plot_psd squeezed to top part of figure

cbrnr, 2019-08-19:
When using `raw.plot_psd` on [this particular GDF file](https://github.com/mne-tools/mne-python/files/3516275/sample_nan.zip) (please unzip first), I get this strange figure:
[image]

So far, this is the only file I found to produce this behavior.

```
import mne
raw = mne.io.read_raw_gdf("sample_nan.gdf", preload=True)
raw.plot_psd()
```

larsoner, 2019-08-19:
This should have been fixed by https://github.com/mne-tools/mne-python/pull/6625, assuming you seed this behavior on current `master`, can you see if it worked okay before that PR?

cbrnr, 2019-08-20:
Yep, that fixed it and it wasn't working before.

closed 2019-08-20 by cbrnr