mne-tools/mne-python #6685: Select channels for dipole fit

Norbsis, 2019-08-21:
I have a problem selecting channels for dipole fitting. Is there a way to select channels by using MNE Python? With Neuromag software it is possible to select channels, but haven't succeeded with MNE Python. 

I use the fit_dipole function and give it an Evoked object with only selected 90 channels and noise_covariance made from Epochs including only those same 90 channels. As a result i get a dipole fitted totally wrong place. Results are better without channel selection. What am I doing wrong?

`
epochs_ch = epochs.copy().pick_types(selection=channels)
noise_cov = mne.compute_covariance(epochs_ch, tmax = 0.0)
evoked_ch = evoked.copy().pick_channels(channels)

dipoles, _ = mne.fit_dipole(evoked_ch.copy().crop(t_max, t_max), noise_cov, fname_bem, fname_trans)
`

agramfort, 2019-08-21:
how did you pick the 90 channels?

>

Norbsis, 2019-08-21:
They were selected around where the activation was visible (somatosensory area). They are these channels plotted below.
[image]

larsoner, 2019-08-21:
Your code looks fine. Can you see if it works reasonably on the `sample` dataset?

When I do this sort of thing I usually just pick the `evoked` object, no need to pick the `epochs` or `cov` as `fit_dipole` will automatically pick the channels it needs from the `cov`. I've validated it against `Xfit` results and the MNE result matches well (at least when using a diagonal cov, which is all that `Xfit` supports).

drammock, 2019-08-21:
You're crop call uses tmax twice. Is that intentional?

Norbsis, 2019-08-21:
Drammock: Yes, the time is just limited to one specific timepoint. That's why the tmax is there twice.
Larsoner: Okay, I first used the pick just with evoked, but when it didn't work, I tried to limit the cov also. I'll try next with the sample dataset.

Norbsis, 2019-08-21:
Hmm.. the sample dataset seems to work fine (or at least with a left-temporal subset of channels the 'right-auditory' fit is almost the same as without the selection). 

What then could cause the problem with my data? If I select only all the channels from the left side of the brain, the dipole is fitted on the right side close to ear. With all channels the stimulation is correctly fitted on the left side somatosensory area.

agramfort, 2019-08-22:
I doubt there is a bug then. Just use all sensors?

larsoner, 2023-09-25:
I have done this a few times successfully in the last few years so maybe we fixed some bug. Closing from non-response / inability to reproduce but if someone can come up with a minimal example let's reopen and fix!

closed 2023-09-25 by larsoner