mne-tools/mne-python #6686: Avoid almost white colors in plots with spatial_colors=True

cbrnr, 2019-08-21:
As mentioned in #6678, spatial colors are sometimes almost white and therefore very hard to see. I think we should restrict these colors from getting almost white.

jona-sassenhagen, 2019-08-24:
`sns.set_style('grey')`

cbrnr, 2019-08-25:
Is this the problem or the solution?

larsoner, 2023-10-01:
I think we tweaked the spatial color mapping sometime in the last 4 years to make the light yellow less light so I'll close, but feel free to reopen if it's still an issue

closed 2023-10-01 by larsoner

Mojackhak, 2025-06-13:
I’ve also encountered the same issue with almost white colors when using spatial_colors=True in mne==1.9.0.
As pointed out by [bboyth on the MNE Discourse](https://mne.discourse.group/t/channels-colors-in-psd-plot/6572/5), modifying the _rgb function in
.../env/lib/site-packages/mne/viz/evoked.py helped resolve the issue for me too.

Here’s the modified version of _rgb that worked:
```python
def _rgb(x, y, z):
    """Transform x, y, z values into RGB colors."""
    rgb = np.array([x, y, z]).T
    rgb -= np.nanmin(rgb, 0)
    rgb /= np.maximum(np.nanmax(rgb, 0), 1e-16)  # avoid div by zero
    # Modification to avoid invisible colors:
    # Reduce RGB intensity for overly light colors
    rgb[rgb.sum(axis=1) > 2.5] = rgb[rgb.sum(axis=1) > 2.5] - 0.3
    return rgb
```
Would it be possible to consider integrating this fix (or a similar safeguard) into a future release?