mne-tools/mne-python #6687: ENH: Generalized Dataset Fetcher for OpenNeuro.org

sappelhoff, 2019-08-21:
In the context of my Google Summer of Code with @jasmainak and @agramfort , we needed testing data at several points in time.

In many cases, we resorted to using data available from [OpenNeuro.org](https://openneuro.org). The main feature of this data base is that all data are formatted according to the Brain Imaging Data Structure ([BIDS](bids.neuroimaging.io)) --> so in general, these datasets are well usable with [mne-bids](https://github.com/mne-tools/mne-bids)

# Potential Feature for MNE-Python

--> implement a dataset fetching function in the MNE `datasets` sub-package.

We can get data from openneuro in several ways, all of which add optional dependencies to MNE-Python:

1. via [`aws-cli`](https://aws.amazon.com/cli/) 
    1. install --> `pip install awscli`
    1. usage example -->`aws s3 sync --no-sign-request s3://openneuro.org/ds000248 ds000248 --exclude "*" --include "*.json"`
    1. we could use `subprocess.run` to call awscli

2. via [`boto3`](https://github.com/boto/boto3)
    1. install --> `pip install boto3`
    1. usage example --> I have no clue yet, ... it looks unnecessarily complicated (because it's a single package to manage all of aws)

3. via [`datalad`](https://www.datalad.org/)
    1. install --> `pip install datalad`
    1. note: needs `git-annex` installed
    1. usage example, see [data fetcher in mne-study-template](https://github.com/mne-tools/mne-study-template/blob/master/tests/download_test_data.py)

# Benefits

- get easy and nice access to all OpenNeuro datasets (BIDS format, yay)
- replaces `datasets.py` module in MNE-BIDS

agramfort, 2019-08-22:
awscli solution is beautifully simple

drammock, 2021-03-16:
closing (already implemented: https://github.com/mne-tools/mne-python/issues/9078#issuecomment-799688566)

closed 2021-03-16 by drammock