mne-tools/mne-python #6696: mne.grand_average() behaves differently for evoked data thanmne.evoked.grand_average()

hoechenberger, 2019-08-22:
#### Describe the bug
`mne.grand_average()` and `mne.evoked.grand_average()` produce different results on the same data: the. weights used to calculate the average are not normalized in `mne.grand_average()`.

#### Steps and/or code to reproduce
```python
In[5]: mne.grand_average(evokeds).data.max()
Identifying common channels ...
all channels are corresponding, nothing to do.
Out[5]: 7.730973384174128e-05
```
```python
In[6]: mne.evoked.grand_average(evokeds).data.max()
Identifying common channels ...
all channels are corresponding, nothing to do.
Out[6]: 4.294985213430072e-06
```

#### Expected results
`mne.grand_average()`, when operating on a sequence of `Evoked`s, should produce the  same results as `mne.evoked.grand_average()`

#### Actual results
The results differ. `mne.grand_average()` calls `mne.evoked.combine_evoked()`, which doesn't normalize the weights first:
https://github.com/mne-tools/mne-python/blob/1189d1a755f4297dcb7212d053f7e1b0d12fd082/mne/evoked.py#L896

`mne.evoked.grand_average()` also calls `mne.evoked.combine_evoked()`, but explicitly passe normalized weights, producing the correct result:
https://github.com/mne-tools/mne-python/blob/1189d1a755f4297dcb7212d053f7e1b0d12fd082/mne/evoked.py#L831-L832

#### Additional information
I wonder if `mne.grand_average()` calls `mne.evoked.combine_evoked()` directly, instead of going through `mne.evoked.grand_average()`, which takes care of the correct weighting.

This is on `master`.

agramfort, 2019-08-22:
why the heck do we have 2 functions to do this :( :(
one should be deprecated

drammock, 2019-08-22:
@agramfort you can assign me, I'll get to it this afternoon. I'm the one who most recently dealt with grand average code I think.

On August 22, 2019 1:34:38 AM PDT, Alexandre Gramfort <notifications@github.com> wrote:
>why the heck do we have 2 functions to do this :( :(
>one should be deprecated

commit b97cfbe2 (2019-08-27): MRG, FIX: remove grand_average code duplication (#6711)

* closes #6696

* update whats_new
closed 2019-08-27 by drammock