mne-tools/mne-python #6699: Further functions & methods that currently do not work with pathlib.Path objects

hoechenberger, 2019-08-22:
#6682 has already improved support for `str`-like paths,  like `pathlib.Path` objects. I just ran my source analysis stack again and found many more places where currently only strings are supported, and passing `pathlib.Path` objects instead raises  an error. I'm not sure if this will be really helpful to anyone, but I thought sharing the results with you at least wouldn't hurt :)

#### Plotting
- `mne.viz.plot_alignment()`: `trans` param

#### Source. Spaces
- `mne.write_source_spaces()`: `fname` param
- `mne.read_source_spaces()`: `fname` param

#### BEM
- `mne.write_bem_surfaces()`: `fname` param
- `mne.write_bem_solution()`: `fname` param

#### Forward Solution
- `mne.make_forward_solution()`: `src` and `trans` params
- `mne.write_forward_solution()`: `fname` param

#### Noise Covariance Matrix
- `mne.write_cov()`: `fname` param
- `mne.read_cov():`: `fname` param

#### Inverse Operator
- `mne.minimum_norm.write_inverse_operator()`: `fname` param
- `mne.minimum_norm.read_inverse_operator()`: `fname` param

#### Source Estimate
- `mne.SourceEstimate.save()`: `fname` param
- `mne.read_source_estimate()`: `fname` param

closed 2019-08-25 by agramfort

hoechenberger, 2019-08-25:
Thanks @larsoner, #6701 fixes all of the above, except for `mne.make_forward_solution()`, where I still have to cast my. `Path` objects to strings for the `trans` and `src` parameters.

hoechenberger, 2019-08-27:
@larsoner I'm currently working on getting this one fixed.

cbrnr, 2019-08-28:
What about `mne.preprocessing.read_ica`? I didn't see this function in the list, but it also contains `.endswith` on the file name.

hoechenberger, 2019-08-28:
@cbrnr This list only contains the functions that threw errors during my source analysis workflow, _after_ cleaning & epoching the data. Therefore it may very well be possible that there's more things to fix, like the function you just mentioned. I can have a look at this next week!

PR #6701 (2019-08-25): MRG, ENH: Support path-like objects
Closes #6699.

- Unifies `preload`-as-`str` handling in `io/base.py`
- Adds `path-like` to `_validate_types`
- Makes `_validate_types` support nested `str` options (e.g. `(None, 'path-like')`)
- Modernizes a number of `tmpdir` uses (@massich you should be happy :) )