mne-tools/mne-python #6707: Additionnal Command line functions

vferat, 2019-08-23:
Is there any plan to add more command line functions to **mne-python** and/or  **mne-C** ?

It would be interesting to have some commands such as `mne make_forward`, `mne make_sources` or `mne compute_inverse`

Coupled with [nipype](https://github.com/nipy/nipype/tree/master) or any other pipeline engines, this would provide strong automation tools and better interactions with freesurfer, especially for source reconstruction.

agramfort, 2019-08-23:
I don't think there is any plan for this in mne-C but in mne-python I don't
see any objection
to add some. We just need to have someone to write them :)

>

vferat, 2019-08-23:
I've already worked a bit on this .I will try to rework functions to respect mne standart and make a PR, so we can discuss about it

any help is welcome ;)

vferat, 2019-09-18:
My plan was to creates the following functions in addition to `mne make_setup_source_space`
1. mne setup_forward model ( MNE-C: [mne_setup_forward_model](https://www.nmr.mgh.harvard.edu/mne/dev/manual/c_reference.html#id134) )
```
mne setup_forward model or mne make_bem_model
Options:
  --version             show program's version number and exit
  -h, --help            show this help message and exit
  -s SUBJECT, --subject=SUBJECT
                        Subject name (required)
  --model=FILE          Output file name. Use a name <dir>/<name>-bem.fif
  --ico=ICO             The surface ico downsampling to use, e.g. 5=20484, 4=5120, 3=1280. If None, no subsampling is applied.
  --homog=VALUE         Use a single compartment model (brain only) instead a three layer one (scalp, skull, and brain). f this flag is specified, the options --brainc , --skullc , and --scalpc are irrelevant.
  --brainc=VALUE        Defines the brain compartment conductivity. The default value is 0.3 S/m.
  --skullc=VALUE        Defines the skull compartment conductivity. The default value is 0.006 S/m 
  --scalpc=VALUE        Defines the brain compartment conductivity. The default value is 0.3 S/m.
  -d SUBJECTS_DIR, --subjects-dir=SUBJECTS_DIR
                        Subjects directory
  --verbose             Turn on verbose mode.
  -o, --overwrite       to write over existing files

```
2. mne prepare_bem_model ( MNE-C: [mne_prepare_bem_model](https://www.nmr.mgh.harvard.edu/mne/dev/manual/c_reference.html#id185) )
```
 mne prepare_bem_model or mne make_bem_solution
Options:
  --version             show program's version number and exit
  -h, --help            show this help message and exit
  --model=FILE          BEM model to use.
  --bem_sol=FILE        Output file name. Use a name <dir>/<name>-bem-sol.fif
  --verbose             Turn on verbose mode.
  -o, --overwrite       to write over existing files
```
3. mne forward_solution ( MNE-C: [mne_forward_solution](https://www.nmr.mgh.harvard.edu/mne/dev/manual/c_reference.html#id127) )

```
mne make_forward_solution
Options:
  --version             show program's version number and exit
  -h, --help            show this help message and exit
  --fwd=FILE            Output file name. Use a name <dir>/<name>-fwd.fif
  --info=FILE           filename to a Raw, Epochs, or Evoked file with measurement information.
  --bem_sol=FILE        filename to BEM solution file
  --src=FILE            filename to SourceEstimate file
  --trans=FILE          transformation filename. If ending is fif or fif.gz will be assumed to be in FIF format any other ending will be assumed to be a text file with a 4x4 transformation matrix 
  --meg                 Compute the MEG forward solution.
  --eeg                 Compute the EEG forward solution.
  --mindist=MINDIST     Minimum distance of sources from inner skull surface (in mm).
  --ignore_ref          Do not include reference channels in compensation
  -n N_JOBS, --n-jobs=N_JOBS
                        The number of jobs to run in parallel (default 1).
                        Requires the joblib package.
  --verbose             Turn on verbose mode.
  -o, --overwrite       to write over existing files
```
Maybe adding more options to `mne make_forward_solution` to integrate `mne.convert_forward_solution()` or to create a separate command `mne convert_forward_solution`

Including sphere model in` mne forward_solution` seems to need a lot of efforts, as the parameters are quite different between [mne.make_sphere_model](https://www.nmr.mgh.harvard.edu/mne/stable/generated/mne.make_sphere_model.html) and [mne_forward_solution](https://www.nmr.mgh.harvard.edu/mne/dev/manual/c_reference.html#id127) and this would make the command less understandable from my point of view.

Descriptions may change, but does the plan is ok for you ? Do you have any suggestion ?

agramfort, 2019-09-18:
no strong feeling here. but let's do one command at a time in different PRs

agramfort, 2019-12-28:
I think this is done now.

closed 2019-12-28 by agramfort