mne-tools/mne-python #6718: ica.plot_properties: y-axis scale missing in variance plot

SophieHerbst, 2019-08-29:
In mne 18.2., the plot that shows the variance captured by a component over segments (lower right) generated by ica.plot_properties lacks the tick marks (scale) for the y-axis. Could that be brought back? It helps to have an idea of the scale. Thanks!

agramfort, 2019-08-29:
indeed we should not have AU for y-label and scale.

PR welcome

SophieHerbst, 2019-08-29:
There isn't even a scale:
https://mne.tools/stable/auto_tutorials/preprocessing/plot_artifacts_correction_ica.html#sphx-glr-auto-tutorials-preprocessing-plot-artifacts-correction-ica-py

mmagnuski, 2019-09-01:
@agramfort Last time this was discussed we decided variance should be AU, but eliminating the ticks and their values was mostly an aesthetic maneuver IIRC:
https://github.com/mne-tools/mne-python/pull/5748

BTW, I am not against bringing the y ticks back, I was actually not so happy to remove them, but I left it for the others to decide.

cbrnr, 2019-09-02:
I just skimmed through the discussion in #5748 and actually no one wanted the y axis ticks and labels removed - it was a misunderstanding and apparently we forgot this because we were too busy with bikeshedding. I can do a PR unless @SophieHerbst wants to do it.

agramfort, 2019-09-02:
go for it @cbrnr

thx

SophieHerbst, 2019-09-02:
@agramfort, my issue was just about the ticks, so you get an idea of how much the component diverges. AU seems right to me. Thanks!

PR #6729 (2019-09-03): [MRG] Bring back yticks for ICA properties
Fixes #6718. In addition to bringing back the yticks and labels, I've removed the space before the colon in the title, and " (index)" from the x axis label (because "segment" is sufficient).

Here's what it looks like after the changes:
[image]

I'd also like to change the horizontal position of the y axis label for the ERP/ERF - the "AU" label is too far to the left and should be closer to the actual axis. Anyone know how to do that quickly? Otherwise, this is ready to merge.
closed 2019-09-03 by larsoner