mne-tools/mne-python #6737: visualization of autoreject results in MNE

jasmainak, 2019-09-04:
Currently, the `autoreject` code hacks into MNE by copying the epochs plotting code and adding [two parameters](https://github.com/autoreject/autoreject/issues/153). Unfortunately this is a bit unwieldy to maintain on the `autoreject` side as we're using a ton of private functions in MNE and duplicating almost 1000 lines of code ... 

are people open to adding these parameters to `epochs.plot` in MNE, or alternatively some code to interpret the `drop_log` in a special way if it starts with the keyword `AUTOREJECT`. I can make a PR with the existing diff and iterate the API from there if people are okay with this.

drammock, 2019-09-04:
no objection from me.  I can work on this with you at the upcoming sprint if you want...  it may have been my recent changes to the plotting code that caused this in the first place

larsoner, 2019-09-05:
- the first param, `bad_epoch_idx` -- I'm not sure why we'd need it, can't you just do `epochs[bad_epoch_idx].plot()`? If not, can you describe what it's trying to do?
-  the second param `log_labels` -- it seems like its job is to color specific parts of the `(n_channels, n_epochs)` plotting grid. So maybe the thing to do is make a new param called `epoch_colors` that defaults to `'auto'`, but can be any `ndarray` of `str` that broadcasts to shape `(n_channels, n_epochs)`.
    - Each entry in the array is expected to be an `mpl-like` color string. (This is general enough to support arbitrary colors via the mpl hex strings, even if it's a bit cumbersome).   - if people want to color specific channels they can pass `(n_channels,1)` array
   - if they want to color squares in the plane (like you do) they pass an array of shape `(n_channels, n_epochs)`.
   - if they want to color specific epochs the can pass `(n_epochs,)` array

I think this would be more general than just what's required for `autoreject` without adding too much more work.

jasmainak, 2019-09-05:
> can't you just do epochs[bad_epoch_idx].plot()

you could but I think it's nicer for people to see them next to the good epochs in red as you see them when you mark them interactively

I'm fine making `log_labels` more general than `autoreject`. I guess we could just make it so that the horizontal scrollbar colors the epoch if a column in `epoch_colors` have the same repeated value. Wdyt?

larsoner, 2019-09-05:
> you could but I think it's nicer for people to see them next to the good epochs in red as you see them when you mark them interactively

What does `bad_epoch_idx` actually do, then? If all it does is change the color, then that would already be handled by the proposed `epoch_colors` arg.

> I guess we could just make it so that the horizontal scrollbar colors the epoch if a column in epoch_colors have the same repeated value. Wdyt?

Whatever ends up looking reasonable. An average over the colors for the given epoch might be fine, too.

jasmainak, 2019-09-05:
> What does bad_epoch_idx actually do, then? If all it does is change the color, then that would already be handled by the proposed epoch_colors arg.

yep I think so

PR #6855 (2019-10-05): MRG: allow users to annotate epochs with colors
closes #6737 

just started by copying over changes from autoreject. Will iterate tomorrow
closed 2019-10-05 by agramfort