mne-tools/mne-python #6740: FFT without adjusting wave length when performing CWT

uesseu, 2019-09-09:
#### Problem
FFT of scipy does not require wave whose length is 2^n.
But, mne.time_frequency.trf._cwt forces make wave length 2^n.

#### My solution
I think, FFT of scipy factorizes wave length (if possible, to prime number).
The factorizing code is in scipy/fft/_pocketfft/pocketfft_hdronly.h line 1242.
It is known that, If wave length can be factorized, (elegant, but little slower)
new FFT algorithms can be applied, and zero padding is not necessary.
Code of _cwt seems to intends to use older FFT algorithm.
If there is a option to avoid zero padding, I want to consider to use it.

#### Additional context
Path of document is this.
```bash
scipy/fft/_pocketfft/README.md
```

And, there is a factorization code in this file. Line 1242.
```bash
scipy/fft/_pocketfft/pocketfft_hdronly.h 
```

agramfort, 2019-09-09:
we could offer this option but I am curious to know if you expect faster
computation
or better results?

>

uesseu, 2019-09-09:
I am sorry, it does not return better results at all.

closed 2019-09-09 by uesseu