mne-tools/mne-python #6746: conda environment file name=base

drammock, 2019-09-09:
looks like #6540 changed the `name` field in the environment file from `mne` to `base`.  Was that intentional?  I am -1 on our install instructions changing people's base environment.

larsoner, 2019-09-09:
Yes it was intentional, the idea being that naive users don't want to mess with environments and switching (think about anaconda windows with a Spyder shortcut), and expert users can override it easily

drammock, 2019-09-10:
hmm. OK. it looks like the docs [here](https://mne.tools/dev/install/mne_python.html#installing-mne-python-and-its-dependencies) got updated to reflect this, but [the advanced install docs](https://mne.tools/dev/install/advanced.html#using-the-development-version-of-mne-python-latest-master) still refer to "the `mne` environment". We should fix that.

larsoner, 2019-09-10:
Agreed, feel free to make a PR

commit dbed4f6f (2019-09-10): closes #6746 (#6747)
closed 2019-09-10 by larsoner