mne-tools/mne-python #6770: BUG: Somato BIDS dataset derivatives need to be updated

larsoner, 2019-09-15:
Turns out there was a bug with the conversion in #6414 conversion. The subject used to be `somato` but now it's `01` and the `-src.fif` was never updated. We thus end up with:
```
>>> fwd['src'][0]['subject_his_id']
'somato'
```
which is wrong (and is now arguably correctly treated as an error downstream later by MNE). I have a quick fix for this but it would be good @jasmainak @sappelhoff if someone could re-create the src, fwd, and (if applicable) inv for this subject because they will all have the old, incorrect `subject_his_id` value.

jasmainak, 2019-09-17:
@larsoner how urgent is this? can this wait till the sprint? @sappelhoff any chance you can fix it in the coming few days?

sappelhoff, 2019-09-17:
Unfortunately I am completely swamped and won't be able to submit a fix for this :(

larsoner, 2019-09-17:
It can wait until 0.20, I pushed a workaround in #6771

wmvanvliet, 2019-10-29:
Also, `dataset_description.json` needs to be updated with a new link to relevant docs:

`http://martinos.org/mne/stable/manual/datasets_index.html?#somatosensory` ->
`https://mne.tools/stable/overview/datasets_index.html#somatosensory`

larsoner, 2020-03-04:
@sappelhoff @jasmainak any chance to get to this in the next week or two to make it into 0.20?

sappelhoff, 2020-03-04:
This is the [conversion script](https://github.com/mne-tools/mne-scripts/blob/master/bids_conversion/convert_somato_data.py).

If I understand you correctly, the error is with these lines:

https://github.com/mne-tools/mne-scripts/blob/9e4c77b724bb04a6d20e93720b5dd6cf110ff8eb/bids_conversion/convert_somato_data.py#L227-L231

where we simply rename the existing `fwd` file, instead of recomputing it. As far as I see it, there are no `src` or `inv` files.

Could a fix be as simple as loading the `fwd`, changing `subject_his_id`, and saving under a new name?

agramfort, 2020-03-04:
yes should be as simple as this and then reupload to OSF.

>

larsoner, 2020-03-04:
> Could a fix be as simple as loading the fwd, changing subject_his_id, and saving under a new name?

I would just overwrite the old one with the correct one

sappelhoff, 2020-03-04:
@larsoner to create a new (correct) one, I assume that I would use [`mne.make_forward_solution`](https://mne.tools/stable/generated/mne.make_forward_solution.html)

What should I supply for the `src` and `bem` parameters?

These are all the files I have (printed up to depth 3):

```
|MNE-somato-data/
|--- subjects/
|------ somato/
|--------- scripts/
|--------- stats/
|--------- label/
|--------- touch/
|--------- surf/
|--------- mri/
|--------- trash/
|--------- tmp/
|--------- src/
|--------- bem/
|--- MEG/
|------ somato/
|--------- sef_raw_sss-trans.fif
|--------- somato-meg-oct-6-fwd.fif
|--------- sef_raw_sss.fif

```

PR #7387 (2020-03-04): Fix somato dataset
closes #6770 

see also: https://github.com/mne-tools/mne-scripts/pull/22

This should do the trick :+1: thanks for guidance @larsoner
closed 2020-03-04 by larsoner