mne-tools/mne-python #6782: Better handling of montage files

jasmainak, 2019-09-17:
When reading montage files, users seem to face a number of issues.

1. Documentation of the different file formats is not clear. For example, *.hpts should not have a header and a column of type. This is not clearly explained anywhere. You cannot figure this out without looking at the code.

2. Inconsistent handling of `numpy` versions when reading text files. Here it is handled properly: https://github.com/mne-tools/mne-python/blob/master/mne/channels/montage.py#L325-L328 but this is not the case in all calls. Ideally, this should be abstracted away into a private helper function.

maybe an easy issue for @fmamashli to solve who encountered this in the first place

agramfort, 2019-09-18:
FYI the read_montage is currently being deprecated in
https://github.com/mne-tools/mne-python/pull/6764

to make sure we are addressing the issue @fmamashli
<https://github.com/fmamashli> can you clarify the usecase? How was the
.hpts
file produced? what does it contain? points in what coordinate system?

>

massich, 2019-09-18:
we bumped the minimum version of numpy. We no longer need to keep both versions of skip_row / header.

fmamashli, 2019-09-18:
We digitized the channels using a new system: [Localite](https://www.localite.de/en/products/tms-navigator/). 
.hpts: we followed MNE manual: EEG, channels name, coordinates (x,y,z) in mm.
The coordinate system is RAS. 

The problem was that in MNE-python, there was no good documentation on the format of the files and what exactly it should contain. like headers, columns, coordinate system.

jasmainak, 2019-09-18:
also just to add to what @fmamashli said, see the [documentation of mne-c](https://mne.tools/stable/manual/c_reference.html?highlight=hpts#mne-convert-dig-data). It is clearly mentioned what the format should look like.

agramfort, 2019-09-18:
if you use the current master you can try this command:
https://mne.tools/dev/generated/mne.channels.make_dig_montage.html#mne.channels.make_dig_montage

provided you have the locations already read in python

you could ideally specify that the points are in MRI RAS and then it should
work just fine (hopefully).

share some fake data if needed.

>

jasmainak, 2019-09-22:
We managed for now. But I would say you need a documentation page analogous to [this one](https://www.nmr.mgh.harvard.edu/mne/stable/manual/io.html) for reading in raw files.

agramfort, 2019-09-23:
should be fine now with:

https://mne.tools/dev/auto_tutorials/misc/plot_sensor_locations.html#sphx-glr-auto-tutorials-misc-plot-sensor-locations-py

https://mne.tools/dev/overview/implementation.html#dig-formats

closed 2019-09-23 by larsoner

jasmainak, 2019-09-23:
Cool !