mne-tools/mne-python #6784: ENH: API for subcortical surface source spaces

larsoner, 2019-09-18:
It turns out (I think!) you can take just about any subcortical ROI and turn it into a surface using a few FreeSurfer commands:
```
mri_tessellate aseg.mgz 17 lh.hippocampus.orig
mri_tessellate aseg.mgz 53 rh.hippocampus.orig
mris_extract_main_component lh.hippocampus.orig lh.hippocampus.orig
mris_extract_main_component rh.hippocampus.orig rh.hippocampus.orig
mris_smooth -nw lh.hippocampus.orig lh.hippocampus.smooth
mris_smooth -nw rh.hippocampus.orig rh.hippocampus.smooth
```
The `.smooth` surfaces look reasonable for the MNE `sample` subject:

[image]

And they can be inflated to a sphere (so that they can be subdivided via ico/oct) with:
```
mris_inflate -no-save-sulc lh.hippocampus.smooth lh.hippocampus.inflated
mris_inflate -no-save-sulc rh.hippocampus.smooth rh.hippocampus.inflated
mris_sphere -q lh.hippocampus.inflated lh.hippocampus.qsphere
mris_sphere -q rh.hippocampus.inflated rh.hippocampus.qsphere
```

Here is a starting API proposal:

1. `setup_source_space(..., surf='...')` allow `surf` to be a list/tuple of [names in aseg.mgz](https://surfer.nmr.mgh.harvard.edu/fswiki/FsTutorial/AnatomicalROI/FreeSurferColorLUT), not just `surf/` files. The default would stay `'white'`, which is an alias for (as would be `pial` and other cortical surfaces) `('lh.white', 'rh.white')` to create a cortical surface source space.
2. Currently we use `FIFF.FIFFV_MNE_SURF_LEFT_HEMI = 101` for left, 102 for right; maybe we should use `1000+aseg_value` for any non-cortical surface, and left/right hemi would keep their "special" / exceptional values of 101 and 102 (their numbers in `aseg.mgz` appear to be 2 or 3 and 41 or 42)
3. The `.kind` for a `SourceSpaces` with just these surfaces would be not `'surf'` (keep this for *cortical* surf source spaces only) but instead `'subcortical_surf'` or something. And combinations of cortical surf + subcortical surf would be, as usual, `'mixed'`.
4. Figure out morphing somehow. In theory we might be able to do this procedure for `fsaverage` and use FreeSurfer commands to align the surfaces (make `.sphere` instead of just `.qsphere` surfaces). Now that `make_morph_maps` is not horribly slow anymore, creating the morph maps on the fly might be acceptable since these surfaces in general will have way fewer vertices.

agramfort, 2019-09-18:
sounds good !

larsoner, 2023-11-03:
From a meeting with @jasmainak it seems like a tentative workable plan is:

1. Leave subcortical segmentation to a separate Python package for now (it's a hard problem that takes a while to run, so a separate tool makes sense)
2. Get that package to output FreeSurfer-compatible `.surf` files to `subjects_dir/subject/...` somewhere (`surf/subcort.lh.*`?)
3. Build source space support for those files using a variant of the API above. Depending on how FreeSurfer-like the outputs are, it might make sense to create a separate `mne.setup_subcortical_source_space`, e.g., if the `spacing` parameter no longer makes sense.
4. Figure out how to support across-subject smoothing using `SourceMorph`

Good next steps would be:

1. @jasmainak share code with me for segmenting cerebellum with me
2. I run it on `sample`
3. Examine outputs
5. Try hacking something with `mne.setup_source_space` from it, then forward, then inverse. Then visualize traces with `matplotlib` for now

dasdiptyajit, 2025-03-22:
@larsoner I am working on a similar problem. If possible have a look at this: 
https://mne.discourse.group/t/morphing-subjects-hippocampus-source-space-to-fsaverage-template/10995