mne-tools/mne-python #6792: DOC, RFC: better visibility for "algorithms" pages

drammock, 2019-09-19:
#6767 yielded a few documentation pages that are live, but are not very findable; they are located in `doc/overview/algorithms`. Most are very good; a couple of them may need some cleanup / fleshing out. Find a way to make them more discoverable. Possibilities:

1. an algorithms TOC page that is linked from `docs/overview/index.rst`
2. combine the algorithms pages into one page that is linked from `docs/overview/index.rst`
3. merge the algorithms with the "implementation details" content
4. other ideas ???

cc @jasmainak

jasmainak, 2019-09-21:
I don't quite like "implementation details" because it gives the impression that they are not important. Something I've realized after discussions with @mshamalainen is that we miss the mathematical details in MNE-Python. 

Scikit-learn does a good job in their "user guide" and so does the MNE-C manual. But MNE-Python really skims over it. I shouldn't have to go read 5 papers to figure out what is the one equation of beamformers.

drammock, 2019-09-22:
looks like in #6767 we also lost [the io page](https://www.nmr.mgh.harvard.edu/mne/stable/manual/io.html), @agramfort might want to restore it?

larsoner, 2019-09-22:
Yeah we should probably restore it for release

larsoner, 2019-09-22:
There was a lot of overlap between the old pages but I didn't mean to remove the content. So yes restore plus deduplicate is the way to go. @drammock feel free to go for it on Monday, we should be able to release Tues or Wed hopefully

PR #6809 (2019-09-24): MRG, DOC: Restore I/O content via new "tutorials"
This PR:

1. Restores the (relevant) content from I/O that disappeared in #6767
2. Unifies our presentation of "how to read data from systems" by making two new "tutorials", one for MEG, one for EEG, that link to how to read data for each system (alongside relevant considerations). Eventually we can maybe actually have some code and plots for this. The tables for readers have gone away as a consequence, as they became redundant with the TOC at the top of these two pages.
3. Unifies how things are treated in `implementation.rst`. Previously some [... cut]
closed 2019-09-24 by larsoner