mne-tools/mne-python #6794: DOC: 2019-09 documentation TODO list

drammock, 2019-09-19:
In preparation for the Sept/Oct code sprint in Woods Hole, here is a list of TODO items for the documentation. This is not a complete list, but hopefully everyone can find one documentation item to work on during the week.

### probably fairly easy? 

- [ ] #6455 need clearer error message when output of `events_from_annotations` (a tuple of events array + event_id dict) is passed to Epochs constructor as if it were just an events array
- [ ] #6321 some missing glossary entries
- [ ] #5630 add a prominent download button to the "installing MNE-C" page
- [ ] Somewhere in our discussion of freesurfer/fsaverage we should link to [this page](https://surfer.nmr.mgh.harvard.edu/fswiki/Buckner40Notes). Probably a good place is on the `fsaverage` entry of the datasets page (hat tip to #3358)
- [ ] ICA tutorial: should apply baseline before `plot_joint` of artifacts

### medium difficulty?

- [ ] #6266 advanced git page
- [ ] #6431 better description (and literature reference) in dipole fitting example
- [ ] #3257 make a screencast of the coregistration GUI in action and add it to the docs (as a tutorial?)
- [ ] #2847 improve docs for the connectivity module
- [ ] #1857 explain SNR estimation better (where?)
- [ ] #3849 sequential application of separate ICAs for EEG/MEG: add to tutorial?

### probably need some discussion / consensus-building first:

- [ ] #6792 better visibility / discoverability for the algorithms pages
- [ ] #6778 the "cookbook" should be either converted to a tutorial, or its contents should get divvied up among existing tutorials / other doc pages 
- [ ] #3669 and #5937 discuss updates to the "flow diagram"
- [ ] #3465 improved docs around transforming between different coord. systems

### earlier versions of checklists like this one (click through if all the good ones above are taken :slightly_smiling_face:):

- [ ] #4188 
- [ ] #3985 
- [ ] #1495

larsoner, 2020-12-01:
@drammock feel free to look through here and mark any you think are needed for 0.22 if we do manage to release in the next two weeks

drammock, 2020-12-01:
none of these are blockers for 0.22 IMO.