mne-tools/mne-python #6796: Can't read eeglab file: AttributeError

psyhbf, 2019-09-20:
Hi,
I am a new mne-python user. when I am trying to import evoked eeglab data for topomap:
import mne
data=mne.read_epochs_eeglab('sub01_epochs_BEFORE_ICA_rej_wrong_extreme_S1_Con.set')

there occurs an error:  
AttributeError: 'Bunch' object has no attribute 'trials'
[image]

I don't know where the problem is. 
Many thanks for any comment.

agramfort, 2019-09-20:
can you share the file with me so I can look at it?

>

psyhbf, 2019-09-20:
Please see the attached file.
thank you very much!

 
 

从QQ邮箱发来的超大附件

sub01_orthogonal_simon_epochs_BEFORE_ICA_rej_wrong_extreme_S1_Con.zip (89.66M, 2019年10月20日 21:45 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?t=exs_ftn_download&k=1f316531cbe250c24a76401a16660717530255090607510848025206064b015e01504809035005155c035c0008070300565453553023354b105355006f09474c0d5e025e5e0759671658085e5e3950480a520d426f24707e2a63206e7925746717540f6e47145a56026e004944145055006e36006f255a564b4b0c41305b&code=e1e10f58

larsoner, 2019-09-20:
Attachments to emails to GitHub (which just make comments on the GitHub site) do not come through. Can you upload to DropBox or email @agramfort separately?

psyhbf, 2019-09-20:
> Attachments to emails to GitHub (which just make comments on the GitHub site) do not come through. Can you upload to DropBox or email @agramfort separately?

Sorry. the file is too big. I have emailed @agramfort.

agramfort, 2019-09-20:
it's a problem with the file (although EEGLAB reads it fine)

you have one level too much of EEG structure.

I need to do:

eeg['EEG']['EEG']

rather than:

eeg['EEG']

how did you save this file?

>

psyhbf, 2019-09-21:
it was saved by "pop_saveset()".
is it possible the using incompatible software? (matlab 7.9.0, eeglab v11.0.5.4b)

agramfort, 2019-09-23:
https://github.com/mne-tools/mne-python/pull/6806/files should fix it. I'll merge when green.

PR #6806 (2019-09-23): fix eeglab nested EEG structure
closes #6796
closed 2019-09-23 by agramfort