mne-tools/mne-python #6801: can't raw.filter MISC channels

ossadtchi, 2019-09-21:
Hello!

With raw.filter I can not filter the instance of the raw created with pick_channels using names of MISC channels. It complains about not finding the channels to work with. Is it intended and one is supposed to filter only EEG or MEG channels? I would love to have a "parallel" code handling the two types of channels!

works:
```
raw_grad = raw.copy().pick_channels(names_grad);
raw_grad.filter(1, 45., h_trans_bandwidth='auto', filter_length ='auto', phase = 'zero');
```
does not work:
```
raw_acc = raw.copy().pick_channels(names_acc);
raw_acc.filter(0.5, 5., h_trans_bandwidth='auto', filter_length ='auto', phase = 'zero');
```
THANKS!!

agramfort, 2019-09-21:
`raw.filter` filters only data channels (EEG, MEG). You need to pass picks parameter manually to the  filter method. Please reopen if it does not work.

ossadtchi, 2019-09-22:
Thanks, Alex, but if I understood you correctly 
neither 
```
raw_misc.filter(0.5, 5., h_trans_bandwidth='auto', filter_length='auto',
                            phase = 'zero', picks = names_misc);
```
nor
```
raw_misc.filter(0.5, 5., h_trans_bandwidth='auto', filter_length='auto',
                             phase = 'zero', picks = np.array([0,1,2])); ```

appear to work and still give the same error. :(

agramfort, 2019-09-22:
can you share a minimal code snippet to replicate the issue using mne sample data?

ossadtchi, 2019-09-22:
Thanks! Yes, here it is. Should be OK with any raw fif file. 
```
tmin, tmax = 0, 50  

raw = mne.io.read_raw_fif(raw_fname) #, add_eeg_ref=False)
raw.crop(tmin, tmax).load_data()

# make a list of names for the grads and misc
picks_grad = mne.pick_types(raw.info, stim=False, meg = 'grad');
names_grad  = [raw.info['ch_names'][n] for n in picks_grad];
picks_misc = mne.pick_types(raw.info, stim=False, meg = False, misc = True);
names_misc  = [raw.info['ch_names'][n] for n in picks_misc];
raw_grad = raw.copy().pick_channels(names_grad);
raw_misc = raw.copy().pick_channels(names_misc);

# works:
raw_grad.filter(1, 45., h_trans_bandwidth='auto', filter_length='auto',
                            phase = 'zero');
# fails:
raw_misc.filter(0.5, 5., h_trans_bandwidth='auto', filter_length='auto',
                           phase = 'zero', picks = names_misc);
# fails:
raw_misc.filter(0.5, 5., h_trans_bandwidth='auto', filter_length='auto',
                           phase = 'zero', picks = np.arange(0,21,1));
```

ossadtchi, 2019-09-22:
There seem to be no MISC channels in sample_audvis_raw.fif. 
```
data_path = op.join(mne.datasets.sample.data_path(), 'MEG', 'sample')
raw = mne.io.read_raw_fif(op.join(data_path, 'sample_audvis_raw.fif'))
picks_misc = mne.pick_types(raw.info, stim=False, meg = False, misc = True);
```
returns empty ```picks_misc```

agramfort, 2019-09-22:
here is another snippet I was hacking

```
import os.path as op

import mne
from mne.datasets import sample

data_path = sample.data_path()
raw_fname = op.join(data_path, 'MEG', 'sample', 'sample_audvis_raw.fif')

tmin, tmax = 0, 50

raw = mne.io.read_raw_fif(raw_fname)
raw.crop(tmin, tmax).load_data()

raw.set_channel_types({'EEG 001': 'misc'})

# make a list of names for the grads and misc
picks_grad = mne.pick_types(raw.info, stim=False, meg='grad')
names_grad = [raw.info['ch_names'][n] for n in picks_grad]
picks_misc = mne.pick_types(raw.info, stim=False, meg=False, misc=True)
names_misc = [raw.info['ch_names'][n] for n in picks_misc]
raw_grad = raw.copy().pick_channels(names_grad)
raw_misc = raw.copy().pick_channels(names_misc)

# works:
raw_grad.filter(1, 45.)
# fails:
raw_misc.filter(0.5, 5., picks=names_misc)
# fails:
raw_misc.filter(0.5, 5., picks=[0])
```

@larsoner this is a regression that appeared with the new picks API.

agramfort, 2019-09-22:
@larsoner I would appreciate your help here as it touches the new logic of pick. I can hack something but you certainly have a simpler approach in mind.

ossadtchi, 2019-09-22:
may be introducing  a special channel type that CAN be filtered but normally there are no channels of this  type. Then, "non-filterable", but requested by the user channels, can be reassigned to this special type, filtered and then set back to their original type.

larsoner, 2019-09-22:
I can look tonight or tomorrow, should be some simple fix in principle

closed 2019-09-23 by larsoner

enricovara, 2021-01-25:
May be related:

`raw_data.resample(Fs, npad='auto', window='boxcar', stim_picks='misc', n_jobs=1, events=None, pad='reflect_limited')`

yields an error:

>   File "<decorator-gen-173>", line 24, in resample
>   File "{envpath}/lib/python3.8/site-packages/mne/io/base.py", line 1166, in resample
>     if len(stim_picks) > 0:
> TypeError: len() of unsized object

`raw_data.resample(Fs, npad='auto', window='boxcar', stim_picks=ch_names, n_jobs=1, events=None, pad='reflect_limited')` where ch_names is a list containing the misc channel names doesn't work either:

>   File "<decorator-gen-173>", line 24, in resample
>   File "{envpath}/lib/python3.8/site-packages/mne/io/base.py", line 1168, in resample
>     data_chunk[stim_picks], n_new, data_chunk.shape[1])
> IndexError: arrays used as indices must be of integer (or boolean) type

A bit new to this apologies if I'm getting it wrong

larsoner, 2021-01-25:
When resampling the `stim` picks *are not low-passed* so this probably does not do what you want (it will lead to aliasing). If you really want this behavior you can pass `stim_picks=pick_types(raw.info, misc=True, exclude=[])` or so.

enricovara, 2021-01-25:
My `stim` is already low-passed but thank you very much for the warning anyway.

Using `stim_picks=pick_types(raw.info, misc=True, exclude=[])` works great (if I do `from mne.io.pick import pick_types` first), thank you again @larsoner !

Might it be worth integrating the above solution by replacing the `stim_picks` argument with `pick` to allow flexibility for e.g.: setups which record stim channels as `misc` rather than `stim`? This would make it more consistent with `raw.filter` (the reason I posted here in the first place) and would avoid the need to call `pick_types` separately.

larsoner, 2021-01-25:
Yes that would be good, basically under the hood `stim_picks` should call `_picks_to_idx` properly and then everything should automagically work, I think. Can you make a PR for this @enricovara ?

PR #6804 (2019-09-23): MRG, BUG: Filter instances with no data channels
Closes #6801.