mne-tools/mne-python #6807: ValueError: Found a key with "/" when saving single tial power

SophieHerbst, 2019-09-23:
I am trying to save single trial power to .h5 and get the following error: 

>   File "/home/sh254795/anaconda3/envs/mne/lib/python3.6/site-packages/mne/externals/h5io/_h5io.py", line 148, in _triage_write
>     where + '["%s"]' % key, cleanup_data=cleanup_data, slash=slash)
> 
>   File "/home/sh254795/anaconda3/envs/mne/lib/python3.6/site-packages/mne/externals/h5io/_h5io.py", line 128, in _triage_write
>     raise ValueError('Found a key with "/", '
> 
> ValueError: Found a key with "/", this is not allowed if slash == error

It seems to come from the event_id, which is a dict: 

> power.event_id
> Out[94]: {'pitch/NP': 1, 'pitch/P': 2, 'time/NP': 3, 'time/P': 4}
> 

I would like to preserve the event information for single trials. How would I turn off the slash='error' in write_hdf5 (and would that cause any trouble?) 
 
Here is my code. I can share the epochs if needed: 
```
power = tfr_multitaper(epochs,freqs=TFR_freqs,n_cycles=n_cycles,
                       time_bandwidth = time_bandwidth,
                       use_fft=True,return_itc=False, decim=2,
                       average = False)

power_savename = '/volatile/PredImEx_Scripts/POWER/' + nip + '_power.h5'
power.save(power_savename, overwrite = True) 
```

Thanks!
Sophie

SophieHerbst, 2019-09-23:
Well, 
`power.event_id = {'pitch_NP': 1, 'pitch_P': 2, 'time_NP': 3, 'time_P': 4}`
fixes it, but is there a way to preserve the event names with 2 conditions?

agramfort, 2019-09-23:
can you share with me a code snippet to replicate the pb? ideally I should
just copy paste
to see the issue. It will save me a bit of time.

thx

>

SophieHerbst, 2019-09-24:
Hi Alex, Here is the code snippet. I am sending you a download link for an example epo.fif
Thank you!

```
import mne 
from mne.time_frequency import tfr_multitaper
import numpy as np

TFR_freqs = np.linspace(.5,40.,num = 40)
n_cycles = np.linspace(2, 10.,num = len(TFR_freqs))
time_bandwidth = 2.

preproc_dir = '/neurospin/meg/meg_tmp/PredImEx_Herbst_2017/data/meg/preproc/'
epochs =  mne.read_epochs(preproc_dir + 'Example-epo.fif', preload=True)  

power = tfr_multitaper(epochs,freqs=TFR_freqs,n_cycles=n_cycles,
                           time_bandwidth = time_bandwidth,
                           use_fft=True,return_itc=False, decim=2,
                           average = False)

power.save(preproc_dir + 'Example-TFR.fif', overwrite = True)
```

agramfort, 2019-09-24:
here is an example that does not require to download any data:

```
import os.path as op

import numpy as np
import mne
from mne.datasets import sample
from mne.time_frequency import tfr_multitaper

data_path = sample.data_path()
raw_fname = op.join(data_path, 'MEG', 'sample', 'sample_audvis_raw.fif')
tmin, tmax = 0, 20
raw = mne.io.read_raw_fif(raw_fname)
raw.crop(tmin, tmax).load_data()
events = mne.find_events(raw)
event_id = {'Aud/Left': 1}

freqs = np.linspace(10., 40., num=4)
epochs = mne.Epochs(raw, events, event_id, tmin=-1., tmax=1.)
power = tfr_multitaper(epochs, freqs=freqs, n_cycles=3,
                       time_bandwidth=2.,
                       use_fft=True, return_itc=False, decim=2,
                       average=False)
power.save('example-tfr.h5', overwrite=True)
```

basically the h5io lib we use to write hdf5 files should escape the strings so `/` becomes `\/`

hopefully something like this should work

SophieHerbst, 2019-10-01:
Hi Alex,
thank you for the example. Weirdly, I still get the error when trying to save as h5, both with the sample and my own data.

>   File "<ipython-input-9-de7d096af679>", line 15, in <module>
>     power.save('example-tfr.h5', overwrite=True)
> 
>   File "/home/sh254795/anaconda3/envs/mne/lib/python3.6/site-packages/mne/time_frequency/tfr.py", line 980, in save
>     write_tfrs(fname, self, overwrite=overwrite)
> 
>   File "/home/sh254795/anaconda3/envs/mne/lib/python3.6/site-packages/mne/time_frequency/tfr.py", line 2258, in write_tfrs
>     write_hdf5(fname, out, overwrite=overwrite, title='mnepython')
> 
>   File "/home/sh254795/anaconda3/envs/mne/lib/python3.6/site-packages/mne/externals/h5io/_h5io.py", line 112, in write_hdf5
>     use_json=use_json)
> 
>   File "/home/sh254795/anaconda3/envs/mne/lib/python3.6/site-packages/mne/externals/h5io/_h5io.py", line 155, in _triage_write
>     where + '[%s]' % vi, cleanup_data=cleanup_data, slash=slash)
> 
>   File "/home/sh254795/anaconda3/envs/mne/lib/python3.6/site-packages/mne/externals/h5io/_h5io.py", line 155, in _triage_write
>     where + '[%s]' % vi, cleanup_data=cleanup_data, slash=slash)
> 
>   File "/home/sh254795/anaconda3/envs/mne/lib/python3.6/site-packages/mne/externals/h5io/_h5io.py", line 148, in _triage_write
>     where + '["%s"]' % key, cleanup_data=cleanup_data, slash=slash)
> 
>   File "/home/sh254795/anaconda3/envs/mne/lib/python3.6/site-packages/mne/externals/h5io/_h5io.py", line 148, in _triage_write
>     where + '["%s"]' % key, cleanup_data=cleanup_data, slash=slash)
> 
>   File "/home/sh254795/anaconda3/envs/mne/lib/python3.6/site-packages/mne/externals/h5io/_h5io.py", line 128, in _triage_write
>     raise ValueError('Found a key with "/", '
> 
> ValueError: Found a key with "/", this is not allowed if slash == error

closed 2019-10-02 by larsoner

agramfort, 2019-10-02:
also backported to maint/0.19

we'll cut a new release at the end of the week

>

PR #6878 (2019-10-02): Fix slash tfrs io
closes #6807