mne-tools/mne-python #6810: RFC: Draft for the new release note v0.19

massich, 2019-09-23:
I put together a release note, please feel free to comment on it (or edit).

--- 

Dear community!

We are very pleased to announce the new 0.18 release of MNE-Python (http://martinos.org/mne/stable/).

A few highlights
================

And we caught and fixed more than 50 bugs!

Notable API changes
===================

For a full list of improvements and API changes see:
http://martinos.org/mne/stable/whats_new.html#version-0-19

To install the latest release the following command should do the job:

    $ pip install --upgrade mne

As usual, we welcome your bug reports, feature requests, critiques, and contributions.

Some links:
- https://github.com/mne-tools/mne-python (code + readme on how to install)
- http://mne.tools/stable/ (full MNE documentation)

Follow us on Twitter for general news (https://twitter.com/mne_news)
and for a regular feed of merged PRs (https://twitter.com/mne_python).

Regards,
The MNE-Python developers

70 people made commits that contributed to this release (in alphabetical order):

* 	Abram Hindle
* 	Achilleas Koutsou
* 	Alexander Kovrig
* 	Alexandre Gramfort
* 	Antoine Gauthier
* 	Britta Westner
* 	Bruno Nicenboim
* 	Burkhard Maess
* 	Chris Bailey
* 	Chris Holdgraf
* 	Christian Brodbeck
* 	Christian Clauss
* 	Clemens Brunner
* 	Cristóbal Moënne-Loccoz
* 	Daniel McCloy
* 	David Haslacher
* 	Denis A. Engemann
* 	Dirk Gütlin
* 	Elizabeth DuPre
* 	Eric Larson
* 	Evgenii Kalenkovich
* 	Fede Raimondo
* 	Guillaume Favelier
* 	Hubert Banville
* 	Ivana Kojcic
* 	Jean-Remi King
* 	Jeff Hanna
* 	Joan Massich
* 	Johannes Kasper
* 	Jon Houck
* 	Jona Sassenhagen
* 	Jose Alanis
* 	Jose C. G. Alanis
* 	José C. G. Alanis
* 	José C. García Alanis
* 	Jussi Nurminen
* 	Kambiz Tavabi
* 	Katarina Slama
* 	Keith Doelling
* 	Kostiantyn Maksymenko
* 	Larry Eisenman
* 	Legrand Nicolas
* 	Lorenz Esch
* 	Luke Bloy
* 	Mainak Jas
* 	Maksymenko Kostiantyn
* 	Marijn van Vliet
* 	Mikolaj Magnuski
* 	Milan Rybář
* 	Nathalie Gayraud
* 	Nikolas Chalas
* 	Oleh Kozynets
* 	Quentin Bertrand
* 	Richard Höchenberger
* 	Robert Seymour
* 	Samuel Deslauriers-Gauthier
* 	Sebastián Castaño
* 	Simon Kern
* 	Stanislas Chambon
* 	Stefan Appelhoff
* 	Stefan Repplinger
* 	Steve Matindi
* 	Teon Brooks
* 	Theodore Papadopoulo
* 	Thomas Donoghue
* 	Thomas Hartmann
* 	Thomas Radman
* 	Eberhard Eich
* 	Joshua J Bear
* 	Paul Roujansky

larsoner, 2019-09-23:
Done with my pass

massich, 2019-09-23:
I did my pas aswell.

drammock, 2019-09-23:
I've done mine.

cbrnr, 2019-09-23:
Done.

closed 2019-09-24 by larsoner