mne-tools/mne-python #6814: can not write inverse operator for brainvision

fmamashli, 2019-09-23:
#### Describe the bug
When I read the vhdr file in MNE-C, it provides an identity matrix for device to head transformation, saved in `raw.info['dev_head_t']`
However, when I read the data in MNE-python, the `raw.info['dev_head_t'] = None`

I could fix this error by creating the transformation matrix as follows:

```py
from mne.transforms import Transform
dev_head_t = Transform('meg', 'head', np.identity(4))
raw.info['dev_head_t'] = dev_head_t
```
#### Steps and/or code to reproduce
The error is reproduced with the following script:

```Python
import mne

import os.path as op
from mne.datasets import fetch_fsaverage
from mne.minimum_norm import make_inverse_operator
from mne.io import read_raw_brainvision
from mne import make_fixed_length_events

subjects_dir = '/autofs/space/tahto_001/users/fahimeh/mne_data'

# Download fsaverage files
fs_dir = fetch_fsaverage(verbose=True)
subjects_dir = op.dirname(fs_dir)

# The files live in:
subject = 'fsaverage'
trans = op.join(op.dirname(mne.__path__[0]), 'mne', 'data', 'fsaverage',
                'fsaverage-trans.fif')
src = op.join(fs_dir, 'bem', 'fsaverage-ico-5-src.fif')
bem = op.join(fs_dir, 'bem', 'fsaverage-5120-5120-5120-bem-sol.fif')

data_dir = op.join(op.dirname(mne.__path__[0]), 'mne', 'io', 'brainvision',
                              'tests', 'data')
vhdr_path = op.join(data_dir, 'test.vhdr')
vmrk_path = op.join(data_dir, 'test.vmrk')
eeg_path = op.join(data_dir, 'test.eeg')
montage = op.join(data_dir, 'test.hpts')

raw = read_raw_brainvision(vhdr_path, montage=montage)
raw.set_eeg_reference(projection=True)  # needed for inverse modeling

events = make_fixed_length_events(raw)
epochs = mne.Epochs(raw, events)
evoked = epochs.average()

fwd = mne.make_forward_solution(raw.info, trans=trans, src=src,
                                bem=bem, eeg=True, mindist=5.0, n_jobs=1)
noise_cov = mne.compute_covariance(epochs, tmin=-0.1)                    

inv = make_inverse_operator(raw.info, fwd, noise_cov, loose=0.2, depth=0.8)
mne.minimum_norm.write_inverse_operator('inv.fif', inv)
```

#### Expected results
The output from the `raw.info['dev_head_t']`

```
<Transform  |  MEG device->head>
[[1. 0. 0. 0.]
 [0. 1. 0. 0.]
 [0. 0. 1. 0.]
 [0. 0. 0. 1.]]
```
#### Actual results
Therefore, when I want to write the inverse operator, it gives this error:

```py
  mne.minimum_norm.write_inverse_operator('inv.fif', inv)
Traceback (most recent call last):

  File "/autofs/space/tahto_001/users/TMS/Python/debug_brainvision_trans.py", line 49, in <module>
    mne.minimum_norm.write_inverse_operator('inv.fif', inv)

  File "</autofs/space/tahto_001/users/TMS/mne-python/mne/externals/decorator.py:decorator-gen-300>", line 2, in write_inverse_operator

  File "/autofs/space/tahto_001/users/TMS/mne-python/mne/utils/_logging.py", line 90, in wrapper
    return function(*args, **kwargs)

  File "/autofs/space/tahto_001/users/TMS/mne-python/mne/minimum_norm/inverse.py", line 344, in write_inverse_operator
    write_forward_meas_info(fid, inv['info'])

  File "/autofs/space/tahto_001/users/TMS/mne-python/mne/forward/forward.py", line 934, in write_forward_meas_info
    raise ValueError('Head<-->sensor transform not found')

ValueError: Head<-->sensor transform not found

```
#### Additional information
Paste the output of `mne.sys_info()` here.

larsoner, 2019-09-23:
I think the behavior of MNE-Python setting `info['dev_head_t'] = None` is actually more correct in this case. There is no MEG<->head to transform so there should be no `dev_head_t`.

For example, `None` can be found straight off of the Neuromag scanner:
```
>>> import mne
>>> mne.io.read_info(mne.datasets.sample.data_path() + '/MEG/sample/ernoise_raw.fif')['dev_head_t'] is None
True
```
But we should probably fix the issue where you can't create and save / load a forward and/or inverse without it. Our check should probably look to see if there are MEG channels and only in that case disallow *creation* of fwd or inv (well before writing).

larsoner, 2019-09-23:
(And we should do this before 0.19 or backport a fix :( )

larsoner, 2019-09-23:
For 0.19 I'll revert to doing `info['dev_head_t'] = Transform('meg', 'head', np.eye(4))` so we can punt for now, and decide later if this should be `None`. I think it should be but MNE-C might need to be updated accordingly, and we can discuss with Matti if it makes sense.

jasmainak, 2019-09-24:
@larsoner if you know that you are working with EEG data, then why does `dev_head_t=None` make more sense that `dev_head_t=np.identity(4)` ? It's brainvision data, so the reader should not set it to `None`

agramfort, 2019-09-24:
this is no device when you have EEG so dev_head_t = None is more correct
for me.

>

massich, 2019-09-24:
@jasmainak why the reader should make that assumption. The reader (as a producer) should read and propagate that the transformation was not there. Then the consumer should be aware that the data can be incomplete and decide to do the right thing.

Then if we decide that this piece of data is required and never missing. Then all readers should populate it and `dev_head_t` other than a valid transformation should crash!

massich, 2019-09-24:
(sorry for clossing, I pushed the wrong button)

jasmainak, 2019-09-24:
well if the program crashes when you run the pipeline, you should also give the solution. Currently there is no documentation for the [Transform class](https://www.nmr.mgh.harvard.edu/mne/stable/generated/mne.transforms.Transform.html#mne.transforms.Transform). It's not used in a single example. Otherwise how do you expect people with Brainvision systems to do source localization?

Also, it's inconsistent with what MNE-C does. Try loading the inverse operator saved from MNE-Python in mne-analyze and it fails to do source localization ...

agramfort, 2019-09-24:
let's be pragmatic here and keep it as it is to avoid errors in other
implementations.

>

larsoner, 2019-09-24:
Okay, let's leave as `np.eye` then

closed 2019-09-24 by larsoner