mne-tools/mne-python #6820: find_edf_events missing annotations for EDF+D, but not EDF+C in 0.18

adam2392, 2019-09-24:
Using MacOSX and conda environment, I downloaded two versions of MNE. Using mne==0.17.2 and 0.18 current version, you take two sets of EDF files (EDF+D and EDF+C). They can be manually converted using edfbrowser. 

For the EDF+D file it works for 0.17.2, but not 0.18. 

Both versions work when the file is in EDF+C format though. Sample code is below. Test data is available upon request (@larsoner has seen the data though)

```
import mne

# in these two runs, only 0.17.2 matches for both files
# in 0.18, only the first annotations gets the full annotation list

if __name__ == '__main__':
    fpath = "./test_edfc.edf"
    raw = mne.io.read_raw_edf(fpath, preload=True)
    annotations = mne.events_from_annotations(raw)
    print(annotations)

    fpath = "./test_edfd.edf"
    raw = mne.io.read_raw_edf(fpath, preload=True)
    annotations = mne.events_from_annotations(raw)
    print(annotations)
```

adam2392, 2019-09-24:
I will try looking at: mne/io/edf/edf.py to see how it's correctly finding the events and then look at differences with respect to version 0.18+.

cbrnr, 2019-09-24:
I don't understand what the problem is. Are the annotations incorrect? Or is it the conversion to events?

larsoner, 2019-09-24:
The annotations are missing, yeah

cbrnr, 2019-09-25:
@adam2392 can you share the two files with me as well?

adam2392, 2019-09-25:
@cbrnr https://gist.github.com/adam2392/8a9b687594eb440a0f6a9824a6aa6293

cbrnr, 2019-09-26:
Something is wrong with both files. EDFbrowser can't read these files - the error mentions that the local patient identification field does not comply to the EDF+ standard.

SigViewer (which uses libbiosig) can load `test_edfc.edf` and shows 28 events (which correspond to what you expected). However, even though SigViewer can also load `test_edfd.edf`, it finds 2342 events - which is not what you expected.

I'm not sure if this should be fixed given that these files are likely invalid EDF files. However, you said that both files worked with 0.17.2, so I'm going to check if we can revert/adapt the behavior.

cbrnr, 2019-10-01:
Let's keep this issue open, I'm not sure if we're currently doing the right thing (even though the linked EDF files seem to be non-standard).

I don't understand what this code is doing:
https://github.com/mne-tools/mne-python/blob/8a4cef803db165e2f8a00ad0f304ab60a79bc6f3/mne/io/edf/edf.py#L128-L132

Apparently, `_read_segment_file` is overloaded in that it directly manipulates the input arguments, but in addition also returns the TAL channel data. In master, `tal_data` is a list of length 14, of which only the zeroth element is used in `_read_annotations_edf`. Interestingly, `tal_data[0]` has shape `(171, 158)`, whereas I expected that to be one-dimensional because there is only one TAL channel. This is also why annotations are incorrectly parsed - in v0.17.2, the input argument to `_read_annotations_edf` was an array of shape `(1, 370036)`, but now it is `(171, 58)`. This explains the difference, but I still don't know why this was changed and why the current master apparently works with other files.

@massich it seems like you authored the relevant code portions, do you know what's going on? Am I misunderstanding how this is supposed to work?

agramfort, 2019-10-01:
can you have a code snippet + a file + an expected output to be able to
debug this?
I assume expected output will come from another open source package

>

cbrnr, 2019-10-01:
The problem is that pyEDFlib, which uses EDFlib, gives an error because at least one field in the header does not conform to the EDF standard. I don't know the exact problem, but it might just be an invalid (e.g. UTF8) character. But we can't compare with that lib because of this error.

Then there is BioSig, which also gives an incorrect number of events for the EDF-D file.

In any case, maybe we shouldn't waste time on this given that two other open source packages either give an error or output an unexpected number of events. Maybe these files are really invalid EDF. On the other hand, MNE 0.17.2 gave the expected number of events for both EDF-C and EDF-D.

agramfort, 2019-10-01:
+1 for low priority (on my side)

>

adam2392, 2019-10-01:
@cbrnr I added into the dropbox on the gist, 2 additional files that are an EDF+D and corresponding EDF+C example of each other. The EDF+C file will be able to be opened w/ edfbrowser. 

Then you can run the same test in MNE 0.17.2 vs 0.18+.

cbrnr, 2019-10-01:
Thanks, I think I already fixed it in #6866 - can you double-check?

larsoner, 2019-10-01:
@adam2392 is it possible for you to make versions of these files as small as possible (e.g. 1MB) that still show the problem? If so then we can incorporate them into our testing data as a regression test. The current size (100MB+) is too big for us to do this

cbrnr, 2019-10-01:
Please also make sure to anonymize any personal information that might be in the file.

larsoner, 2019-10-01:
@adam2392 if you have access to the EEG recording device, it might be easier to do a new, short recording

cbrnr, 2019-10-01:
@larsoner EDFbrowser does support cropping, might as well try that first.

adam2392, 2019-10-01:
So I can crop out an EDF+C file easily. The issue is our EDF+D files come from some clinical collaborators. We probably need that for sure though?

cbrnr, 2019-10-01:
We need EDF-D, yes.

cbrnr, 2019-10-02:
I just noticed that EDFbrowser can't open EDF-D files, so obviously cropping is not possible either.

cbrnr, 2019-10-02:
However, EDFbrowser can convert EDF-D to EDF-C, so this might be a valid workaround if you ever come across another EDF-D that doesn't work in MNE. It really seems like EDF-D is very rarely used and therefore poorly supported by various packages.

larsoner, 2019-10-02:
@cbrnr should we go ahead and merge #6866 then? I guess we can add a test later if we find an appropriate file

cbrnr, 2019-10-02:
Yes, I'd merge without the test file for now.

PR #6866 (2019-10-02): Fix TAL channel parsing for EDF-D files
Fixes #6820. Surprisingly, this doesn't even break a single test, and the behavior is now identical to v0.17.2 for the EDF-D test file.
closed 2019-10-02 by larsoner