mne-tools/mne-python #6824: Deymed TruScan montage file (.ced file)

DominiqueMakowski, 2019-09-25:
Hi guys, firstly, thanks again for mne which allowed me to successfully finish my PhD 🥇🎉 

My new workplace has a 128 channels [TruScan Deymed system](https://deymed.com/truscan-eeg), from which I've been trying to get usable data. 

However, the montage is not supported natively by mne. The customer service sent me a .txt file ([deymed_layout_128.txt](https://github.com/mne-tools/mne-python/files/3651336/deymed_layout_128.txt)), in which there are columns related to channel names, Phi, Theta, and x y z coordinates (attached). They also sent a .ced file, which contains the same info, which I cannot attach (but can send it via email). 

Unfortunately, using the `mne.channels.read_montage("deymed_layout_128", path = ".")` doesn't work: 

```python
mne.channels.read_montage("deymed_layout_128", path = ".")
Traceback (most recent call last):

  File "<ipython-input-22-265da416e197>", line 1, in <module>
    mne.channels.read_montage("deymed_layout_128", path = ".")

  File "C:\Users\Dom\Desktop\WPy-3710\python-3.7.1.amd64\lib\site-packages\mne\channels\montage.py", line 281, in read_montage
    pol = np.deg2rad(data[:, 1].astype(float))

ValueError: could not convert string to float: 'Fp1'
```

How can I read this montage? Thanks again!

agramfort, 2019-09-25:
with new 0.19 version you can use make_dig_montage provided you've read the
locations yourself.

Otherwise look at the code of read_custom_montage to see how to handle
polar coordinates

>

DominiqueMakowski, 2019-09-25:
> provided you've read the locations yourself

What do you mean by that? 

I've looked at the biosemi montages shipped by default (the .txt file) and they have three columns, Site (the name), and then Phi and Theta. If I adapt the Deymed .txt file by only keeping the name, phi and theta columns (i.e., removing the x y z coordinates), and then read using `read_montage`, would that work (and be sufficient for correctly displaying / processing the data)?

agramfort, 2019-09-25:
read_montage is deprecated in 0.19

use read_custom_montage, make_standard_montage or make_dig_montage

have a look for example at
https://github.com/mne-tools/mne-python/blob/master/mne/channels/_standard_montage_utils.py#L86

>

closed 2019-09-30 by DominiqueMakowski

agramfort, 2019-09-30:
looks good !

if you can have the fiducials too that would be great and useful for any
source analysis using
such a template montage.

DominiqueMakowski, 2019-09-30:
@agramfort Thanks! 

May I know if these fiducial points are information that I can obtain from the vendor (i.e., fixed values for the cap) or is it something acquired for each individual subject using, for instance, their MRI?

And out of curiosity, why is the coordinates array specified as yxz (I would have expected xyz)?

agramfort, 2019-09-30:
>
> May I know if these fiducial points are information that I can obtain from
> the vendor (i.e., fixed values for the cap) or is it something acquired for
> each individual subject using, for instance, their MRI?
>

you can get it from template. You can also infer it from where the ears are
on the cap.

see for example
https://mne.tools/stable/auto_tutorials/source-modeling/plot_eeg_no_mri.html
to see how it should look like on fsaverage.

> And out of curiosity, why is the coordinates array specified as yxz (I
> would have expected xyz)?
>

it's the convention of the head coord system in MNE. See:
https://mne.tools/dev/auto_tutorials/source-modeling/plot_source_alignment.html#coordinate-frame-definitions