mne-tools/mne-python #6828: Rounding errors in ICA

christian-oreilly, 2019-09-26:
#### Describe the bug
I get important round-off errors when applying ICA to my recordings. 

#### Steps and/or code to reproduce
The following code demonstrates the issue.  

```Python
import urllib.request
import numpy as np
import mne
from mne.preprocessing import ICA

url_epoch = "https://www.dropbox.com/s/zdsttpc8b96tk1k/test-epo.fif?dl=1"
file_name = "test-epo.fif"

with urllib.request.urlopen(url_epoch) as u:
    data = u.read()
with open(file_name, "wb") as f :
    f.write(data)
epochs_ica = mne.read_epochs(file_name)

ica = ICA(method='fastica', random_state=324234)
ica.fit(epochs_ica)

n_epochs, n_chans, n_samples = epochs_ica.get_data().shape

A = epochs_ica.get_data()[0].reshape((n_chans, n_samples*n_epochs))

A_ = A/ica.pre_whitener_ - ica.pca_mean_[:, None]
unmixing = np.dot(ica.unmixing_matrix_, ica.pca_components_)
mixing = np.dot(ica.pca_components_.T, ica.mixing_matrix_)
print("###############")

### This is a reproduction of what is done in ica._pick_sources(...)
### on the line "proj_mat = np.dot(mixing[:, sel_keep], unmixing[sel_keep, :])"
### Note that in this case the indexing is suppose to doing nothing conceptually. cause 
### It would have guess that the rounding error was due to Numpy returning a copy of 
### the index matrices and that the rounding error was occuring during the copy, but 
### apparently, the rounding source is a bit more fancy than that since using copies do
### not reproduce the error.
W1 = np.dot(mixing[:, np.arange(mixing.shape[1])], unmixing[np.arange(mixing.shape[1]), :])
W2 = np.dot(mixing, unmixing)
W3 = np.dot(mixing.copy(), unmixing.copy())
print(np.mean(np.abs(W1-W2)))
print(np.mean(np.abs(W3-W2)))

W1 = np.dot(mixing.astype(np.longdouble)[:, np.arange(mixing.shape[1])], 
            unmixing.astype(np.longdouble)[np.arange(mixing.shape[1]), :])
W2 = np.dot(mixing.astype(np.longdouble), unmixing.astype(np.longdouble))
print(np.mean(np.abs(W1-W2)))
```
The three prints give me...
```
8.208439146064446e-11
0.0
0.0
```
It could be argued that this error is relatively small (8.21e-11), but it should be null here. Anyway, I would not have reported this issue if this was the biggest error that I found. The problem is that I get much higher. Running the same code with the epochs I have on my PC I get...
```
0.00021674458459267328
0.0
```
I struggle to share it because when I save and reload the epoch, I also get rounding errors so that I can not share the file as I have on my PC. For example, with

```
data1 = epochs[0].get_data()
epochs[0].save("test-epo.fif")
data2 = mne.read_epochs("test-epo.fif")[0].get_data()

print(np.max((data1-data2)/data1))
print(np.min(epochs[0].get_data()), np.max(epochs[0].get_data()))
print(np.min(epochs_ica.get_data()), np.max(epochs_ica.get_data()))
```
I get the following printout:
```
5.897469837235818e-08
-0.00024632105744038185 0.00016047083994182436
-0.00024632105955876317 0.00016047084004895164
```
Again, not such big rounding error (although we should be able to expect better than errors on the 8th significant digit I guess) but because of these, I cannot share the exact data that reproduce the 0.000216 error. I guess the point should nevertheless be clear. 

#### Additional information
mne.sys_info():
```
Platform:      Linux-5.0.0-29-generic-x86_64-with-debian-buster-sid
Python:        3.7.3 (default, Mar 27 2019, 22:11:17)  [GCC 7.3.0]
Executable:    /home/christian/anaconda3/bin/python
CPU:           x86_64: 4 cores
Memory:        7.5 GB

mne:           0.18.2
numpy:         1.16.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.2.1
matplotlib:    3.0.3 {backend=Qt5Agg}

sklearn:       0.20.3
nibabel:       2.5.1
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.9.2}
cupy:          Not found
pandas:        0.24.2
dipy:          Not found
****

cbrnr, 2019-09-26:
It seems like some ICA algorithms are numerically not very stable, see e.g. this issue here: https://github.com/mne-tools/mne-python/issues/4922
Back then we decided that there's not much we can do unfortunately, but I'm open to discussing this issue again.

cbrnr, 2019-09-26:
See also the discussion in #3049.

agramfort, 2019-09-26:
2 things:
- no you should not expect 0 as you shuffle the columns and the order in which you do numerical operations matters. This is particularly true if you use multithread linear algebra (mkl_rt here)
- epochs are saved in float32 to avoid too big files. it's a trade off.

finally on my system that uses openblas I get

```
###############
0.0
0.0
0.0
```

try using openblas or disabling multithreading in mkl.

christian-oreilly, 2019-09-26:
Thanks to both of you for this fast feedback.

> no you should not expect 0 as you shuffle the columns and the order in which you do numerical operations matters. This is particularly true if you use multithread linear algebra (mkl_rt here)

In the example I given and I refered to, the indexing did not shuffle anything. It did nothing conceptually, just like x[range(len(x))] does nothing to x conceptually. That is why I was saying that we should expect x not to change under such an operation.

>  try using openblas or disabling multithreading in mkl.

You were right. Switching to Openblas solved the issue. Now I have no rounding error at all. Thanks for the hint.

larsoner, 2019-09-26:
Sounds like things are settled here so let's go ahead and close.

closed 2019-09-26 by larsoner