mne-tools/mne-python #6833: mlab rendering issue

nithyar89, 2019-09-26:
Hi,

i have a very similar problem to this issue https://github.com/mne-tools/mne-python/issues/5463 , where, mne coreg gives me the coreg window with an empty frame where the 3D display should be. Also, coregistration seems to be doing something when I push buttons, and I can set options etc., but no visualization appears on-screen. Also, the console fills with VTK errors that seem to deal with vtkShaderProgram. My pysurfer plots also do not work (brain=stc.plot() etc.) 
exporting MESA_GL_VERSION_OVERRIDE=3.3 does not help as i think the version is already 3.3

In addition to solving this issue using MESA, i wanted to know how to set up OpenGL to use our NVIDIA graphics card (P100, Tesla) for rendering as well as all Mayavi and Python scripts. 

Here is some information for our server environment. 

OpenGL vendor string: VMware, Inc.
OpenGL renderer string: llvmpipe (LLVM 7.0, 256 bits)
OpenGL core profile version string: 3.3 (Core Profile) Mesa 18.3.4
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 3.3 (Compatibility Profile) Mesa 18.3.4
OpenGL shading language version string: 3.30
OpenGL ES profile version string: OpenGL ES 2.0 Mesa 18.3.4
direct rendering: Yes
    GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer, 
    GLX_EXT_visual_rating, GLX_MESA_copy_sub_buffer, GLX_MESA_query_renderer, 
mne.sys_info()
Qt WebEngine seems to be initialized from a plugin. Please set Qt::AA_ShareOpenGLContexts using QCoreApplication::setAttribute before constructing QGuiApplication.

mne.sys_info()
Platform:      Linux-3.10.0-1062.1.1.el7.x86_64-x86_64-with-redhat-7.7-Maipo
Python:        3.7.4 (default, Aug 13 2019, 20:35:49)  [GCC 7.3.0]
Executable:    /usr/bin/anaconda2/bin/python
CPU:           x86_64: 72 cores
Memory:        503.6 GB
mne:           0.18.2
numpy:         1.16.4 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.3.1
matplotlib:    3.1.0 {backend=Qt5Agg}
sklearn:       0.21.2
nibabel:       2.5.0
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.9.2}
cupy:          Not found
pandas:        0.25.0
dipy:          1.0.0

larsoner, 2019-09-27:
MNE 0.19 was released this week, can you try it? It contains some fixes. In particular the `--simple-rendering` flag might fix things for you

nithyar89, 2019-09-27:
Thanks! We will try getting our NVIDIA GPU to render through virtualGL so we can do hardware accelerated graphics first. If that doesnt work, i could install MNE 0.19. Does the --simple-rendering flag force mayavi to use software rendering using the CPU?

larsoner, 2019-09-27:
No it just uses slightly simpler OpenGL rendering techniques

closed 2019-12-28 by agramfort