mne-tools/mne-python #6834: API: Compat between MNE-C and MNE-Python `-annot.fif`

larsoner, 2019-09-27:
This is not so good for compat:
```
$ cd ~/mne_data/MNE-sample-data/MEG/sample
$ touch sample_audvis_raw-annot.fif  # also one saved by MNE-Python will have this happen...
$ mne_browse_raw --raw sample_audvis_raw.fif
...
Annotation file /home/larsoner/mne_data/MNE-sample-data/MEG/sample/sample_audvis_raw-annot.fif deleted
$ mne_browse_raw --raw sample_audvis_raw.fif
... windows -> show annotator -> click around a bit ...
Saved 5 user-defined events to annotation file /home/larsoner/mne_data/MNE-sample-data/MEG/sample/sample_audvis_raw-annot.fif
$ mne what sample_audvis_raw-annot.fif
events
$ mne show_fiff sample_audvis_raw-annot.fif
999 = FIFFB_ROOT
    100 = FIFF_FILE_ID (20b ids) = {'version': 65539, 'machid': a ... dict len=4
    101 = FIFF_DIR_POINTER (4b >i4) = [-1]
    106 = FIFF_FREE_LIST (4b >i4) = [-1]
    108 = FIFF_NOP (0b nul)
    361 = FIFFB_MNE_EVENTS
        3507 = FIFF_MNE_CH_NAME_LIST (11b str) = Annotations ... str len=11
        3561 = FIFF_MNE_EVENT_LIST (60b >u4) = [27317     0  1001 27473     0 ... array size=15
        3566 = FIFF_MNE_EVENT_COMMENTS (120b >B1)>>>>>>>>>>>>>>>>>>>>BAD
```
The only difference between this file and a standard `-eve.fif` is this line:
```
        3566 = FIFF_MNE_EVENT_COMMENTS (120b >B1)>>>>>>>>>>>>>>>>>>>>BAD
```
If I fix our I/O function to give us something nicer here than the `>>>>BAD` error it turns out this is:
```
        3566 = FIFF_MNE_EVENT_COMMENTS (120b bytestr) = Comment # 1 is optionalCommen ... str len=120
```
So in principle these could be read as annotations with zero duration, or as events + descriptions.

Not sure what the best approach would be. But a good first step might be to have MNE-C not nuke our `-annot.fif` files if it finds them.

agramfort, 2020-01-05:
can we close this? if not what do you suggest @larsoner ?

larsoner, 2020-01-05:
I think we need to decide how to handle this. At the very least, MNE-C deleting the file is problematic

agramfort, 2020-01-05:
this means changing mne-c code and nobody will update its version... it's
also now safe maybe to
think that mne python users don't use the C code anymore?

>

larsoner, 2020-01-05:
Okay let's close then

closed 2020-01-05 by larsoner