mne-tools/mne-python #6836: mne.preprocessing.corrmap auto threshold error handling

nugenta, 2019-09-27:
using the threshold="auto" option in corrmap sets the lower threshold for correlation at 0.6 - if there is no correlation above that threshold, the routine dies with the not terribly informative "ValueError: x or y has zero length". Could either have a lower min threshold that's checked, or just provide feedback that there are no maps correlating above threshold. 

Detailed instructions on how to file a bug can be found in our [FAQ](https://mne.tools/dev/overview/faq.html#i-think-i-found-a-bug-what-do-i-do).

If your issue is a usage question, please consider asking on our [Gitter channel](https://gitter.im/mne-tools/mne-python) or on our [mailing list](https://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis) instead of opening an issue.

#### Describe the bug
Please provide a clear and concise description of the bug.

#### Steps and/or code to reproduce
Please provide a code snippet or [minimal working example (MWE)](https://en.wikipedia.org/wiki/Minimal_Working_Example)
to replicate your problem.

This MWE should be self-contained, which means that other MNE-Python contributors
should be able to copy and paste the provided snippet and replicate the bug.
If possible, use MNE-Python testing examples to reproduce the error. Otherwise,
provide a small and anonymized portion of your data required to reproduce the bug.

If the code is too long, feel free to put it in a [public gist](https://gist.github.com) and link
it in the issue.

Example:

```Python
import mne

fname = mne.datasets.sample.data_path() + '/MEG/sample/sample_audvis_raw.fif'
raw = mne.io.read_raw_fif(fname)
raw = mne.concatenate_raws([raw])
raw.save('test_raw.fif', overwrite=True)
raw_read = mne.io.read_raw_fif('test_raw.fif')  # this breaks
```

#### Expected results
Provide a clear and concise description of what you expected to happen.

#### Actual results
Please paste or specifically describe the actual output or traceback.

#### Additional information
Paste the output of `mne.sys_info()` here.

PalAditya, 2019-10-02:
Hi, I am interested in contributing to this awesome project. Could this issue be assigned to me please?

fmamashli, 2019-10-02:
@nugenta I tried to replicate the error using the following script but could not:

```
import os
import mne
from mne.preprocessing import (ICA, create_eog_epochs, create_ecg_epochs,
                               corrmap)
import numpy as np

mapping = {
    'Fc5.': 'FC5', 'Fc3.': 'FC3', 'Fc1.': 'FC1', 'Fcz.': 'FCz', 'Fc2.': 'FC2',
    'Fc4.': 'FC4', 'Fc6.': 'FC6', 'C5..': 'C5', 'C3..': 'C3', 'C1..': 'C1',
    'Cz..': 'Cz', 'C2..': 'C2', 'C4..': 'C4', 'C6..': 'C6', 'Cp5.': 'CP5',
    'Cp3.': 'CP3', 'Cp1.': 'CP1', 'Cpz.': 'CPz', 'Cp2.': 'CP2', 'Cp4.': 'CP4',
    'Cp6.': 'CP6', 'Fp1.': 'Fp1', 'Fpz.': 'Fpz', 'Fp2.': 'Fp2', 'Af7.': 'AF7',
    'Af3.': 'AF3', 'Afz.': 'AFz', 'Af4.': 'AF4', 'Af8.': 'AF8', 'F7..': 'F7',
    'F5..': 'F5', 'F3..': 'F3', 'F1..': 'F1', 'Fz..': 'Fz', 'F2..': 'F2',
    'F4..': 'F4', 'F6..': 'F6', 'F8..': 'F8', 'Ft7.': 'FT7', 'Ft8.': 'FT8',
    'T7..': 'T7', 'T8..': 'T8', 'T9..': 'T9', 'T10.': 'T10', 'Tp7.': 'TP7',
    'Tp8.': 'TP8', 'P7..': 'P7', 'P5..': 'P5', 'P3..': 'P3', 'P1..': 'P1',
    'Pz..': 'Pz', 'P2..': 'P2', 'P4..': 'P4', 'P6..': 'P6', 'P8..': 'P8',
    'Po7.': 'PO7', 'Po3.': 'PO3', 'Poz.': 'POz', 'Po4.': 'PO4', 'Po8.': 'PO8',
    'O1..': 'O1', 'Oz..': 'Oz', 'O2..': 'O2', 'Iz..': 'Iz'
}

raws = list()
icas = list()

for subj in range(4):
    # EEGBCI subjects are 1-indexed; run 3 is a left/right hand movement task
    fname = mne.datasets.eegbci.load_data(subj + 1, runs=[3])[0]
    raw = mne.io.read_raw_edf(fname)
    # remove trailing `.` from channel names so we can set montage
    raw.rename_channels(mapping)
    raw.set_montage('standard_1005')
    # fit ICA
    ica = ICA(n_components=3, random_state=5)
    ica.fit(raw)
    if subj==0:
        ica.pca_components_[:, :] = np.random.rand(64, 64)
        
    raws.append(raw)
    icas.append(ica)

corrmap(icas, template=(0, 0), threshold='auto',plot=False)

```
can you send us a minimal example to regenerate your error?

larsoner, 2019-10-02:
@PalAditya it looks like @fmamashli was also looking, but feel free to give it a try if you'd like

fmamashli, 2019-10-02:
@PalAditya I could not regenerate the error after several try. I asked the person to send a code which does that.

nugenta, 2019-10-02:
I can indeed run your code, but here is the error I get when I try to run this on my data. If I change “auto” to 0.5 in the initial call, corrmap runs.

Here’s the error I get:
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-23-f0a981905c23> in <module>
      4 mne.preprocessing.corrmap([ica], ecg_temp2_short, ch_type="mag",threshold="auto",label='ecg',plot=False)
      5 mne.preprocessing.corrmap([ica], ecg_temp3_short, ch_type="mag",threshold="auto",label='ecg',plot=False)
----> 6 mne.preprocessing.corrmap([ica], eog_temp1_short, ch_type="mag",threshold="auto",label='eog',plot=False)
      7 mne.preprocessing.corrmap([ica], eog_temp2_short, ch_type="mag",threshold="auto",label='eog',plot=False)

</anaconda3/envs/mne_current/lib/python3.6/site-packages/mne/externals/decorator.py:decorator-gen-387> in corrmap(icas, template, threshold, label, ch_type, plot, show, verbose, outlines, layout, sensors, contours, cmap)

/anaconda3/envs/mne_current/lib/python3.6/site-packages/mne/utils/_logging.py in wrapper(*args, **kwargs)
     87             with use_log_level(verbose_level):
     88                 return function(*args, **kwargs)
---> 89         return function(*args, **kwargs)
     90     return FunctionMaker.create(
     91         function, 'return decfunc(%(signature)s)',

/anaconda3/envs/mne_current/lib/python3.6/site-packages/mne/preprocessing/ica.py in corrmap(icas, template, threshold, label, ch_type, plot, show, verbose, outlines, layout, sensors, contours, cmap)
   2583         nt, mt, s, mx = _find_max_corrs(all_maps, nt, threshold)
   2584     elif len(threshold) > 1:
-> 2585         paths = [_find_max_corrs(all_maps, nt, t) for t in threshold]
   2586         # find iteration with highest avg correlation with target
   2587         nt, mt, s, mx = paths[np.argmax([path[1] for path in paths])]

/anaconda3/envs/mne_current/lib/python3.6/site-packages/mne/preprocessing/ica.py in <listcomp>(.0)
   2583         nt, mt, s, mx = _find_max_corrs(all_maps, nt, threshold)
   2584     elif len(threshold) > 1:
-> 2585         paths = [_find_max_corrs(all_maps, nt, t) for t in threshold]
   2586         # find iteration with highest avg correlation with target
   2587         nt, mt, s, mx = paths[np.argmax([path[1] for path in paths])]

/anaconda3/envs/mne_current/lib/python3.6/site-packages/mne/preprocessing/ica.py in _find_max_corrs(all_maps, target, threshold)
   2387 def _find_max_corrs(all_maps, target, threshold):
   2388     """Compute correlations between template and target components."""
-> 2389     all_corrs = [compute_corr(target, subj.T) for subj in all_maps]
   2390     abs_corrs = [np.abs(a) for a in all_corrs]
   2391     corr_polarities = [np.sign(a) for a in all_corrs]

/anaconda3/envs/mne_current/lib/python3.6/site-packages/mne/preprocessing/ica.py in <listcomp>(.0)
   2387 def _find_max_corrs(all_maps, target, threshold):
   2388     """Compute correlations between template and target components."""
-> 2389     all_corrs = [compute_corr(target, subj.T) for subj in all_maps]
   2390     abs_corrs = [np.abs(a) for a in all_corrs]
   2391     corr_polarities = [np.sign(a) for a in all_corrs]

/anaconda3/envs/mne_current/lib/python3.6/site-packages/mne/utils/numerics.py in compute_corr(x, y)
    245     """Compute pearson correlations between a vector and a matrix."""
    246     if len(x) == 0 or len(y) == 0:
--> 247         raise ValueError('x or y has zero length')
    248     X = np.array(x, float)
    249     Y = np.array(y, float)

ValueError: x or y has zero length

Allison C. Nugent, PhD
Director, MEG Core Facility
NIMH/NIH/DHHS
10 Center Drive, MSC 1059
Building 10 Room 4N242
Bethesda, MD 20892-1059
301-451-8863
nugenta@nih.gov<mailto:nugenta@nih.gov>

fmamashli, 2019-10-02:
I have also tested with threshold = 0.5. Can you check the version of your mne: mne.__version__
My guess is that your mne is not updated to the latest version.

nugenta, 2019-10-02:
0.18.2

Allison C. Nugent, PhD
Director, MEG Core Facility
NIMH/NIH/DHHS
10 Center Drive, MSC 1059
Building 10 Room 4N242
Bethesda, MD 20892-1059
301-451-8863
nugenta@nih.gov<mailto:nugenta@nih.gov>

larsoner, 2019-10-02:
@nugenta based on the traceback I think that `target` must be empty/length zero, and that comes from the `template` argument. What are you passing in your use case?

larsoner, 2019-10-02:
The other possibility is that `all_maps = [ica.get_components().T for ica in icas]` is somehow creating empty objects but that would be weird / another error. But it's also possible!

nugenta, 2019-10-02:
Sorry, I think I sent you the wrong traceback. Let me find a dataset that invokes the correct one…

Allison C. Nugent, PhD
Director, MEG Core Facility
NIMH/NIH/DHHS
10 Center Drive, MSC 1059
Building 10 Room 4N242
Bethesda, MD 20892-1059
301-451-8863
nugenta@nih.gov<mailto:nugenta@nih.gov>

nugenta, 2019-10-02:
Sigh. I thought I had found where in the code the issue originated from, but I agree that based on this error, it looks like it’s the template or ica maps at fault.

But this:

mne.preprocessing.corrmap([ica], eog_temp1_short, ch_type="mag",threshold=0.5,label='eog',plot=False)

produces

Median correlation with constructed map: 0.606
At least 1 IC detected for each subject.

And this

mne.preprocessing.corrmap([ica], eog_temp1_short, ch_type="mag",threshold="auto",label='eog',plot=False)

produces this:

ValueError                                Traceback (most recent call last)
<ipython-input-18-97d74457baab> in <module>
----> 1 mne.preprocessing.corrmap([ica], eog_temp1_short, ch_type="mag",threshold="auto",label='eog',plot=False)

</anaconda3/envs/mne_current/lib/python3.6/site-packages/mne/externals/decorator.py:decorator-gen-387> in corrmap(icas, template, threshold, label, ch_type, plot, show, verbose, outlines, layout, sensors, contours, cmap)

/anaconda3/envs/mne_current/lib/python3.6/site-packages/mne/utils/_logging.py in wrapper(*args, **kwargs)
     87             with use_log_level(verbose_level):
     88                 return function(*args, **kwargs)
---> 89         return function(*args, **kwargs)
     90     return FunctionMaker.create(
     91         function, 'return decfunc(%(signature)s)',

/anaconda3/envs/mne_current/lib/python3.6/site-packages/mne/preprocessing/ica.py in corrmap(icas, template, threshold, label, ch_type, plot, show, verbose, outlines, layout, sensors, contours, cmap)
   2583         nt, mt, s, mx = _find_max_corrs(all_maps, nt, threshold)
   2584     elif len(threshold) > 1:
-> 2585         paths = [_find_max_corrs(all_maps, nt, t) for t in threshold]
   2586         # find iteration with highest avg correlation with target
   2587         nt, mt, s, mx = paths[np.argmax([path[1] for path in paths])]

/anaconda3/envs/mne_current/lib/python3.6/site-packages/mne/preprocessing/ica.py in <listcomp>(.0)
   2583         nt, mt, s, mx = _find_max_corrs(all_maps, nt, threshold)
   2584     elif len(threshold) > 1:
-> 2585         paths = [_find_max_corrs(all_maps, nt, t) for t in threshold]
   2586         # find iteration with highest avg correlation with target
   2587         nt, mt, s, mx = paths[np.argmax([path[1] for path in paths])]

/anaconda3/envs/mne_current/lib/python3.6/site-packages/mne/preprocessing/ica.py in _find_max_corrs(all_maps, target, threshold)
   2387 def _find_max_corrs(all_maps, target, threshold):
   2388     """Compute correlations between template and target components."""
-> 2389     all_corrs = [compute_corr(target, subj.T) for subj in all_maps]
   2390     abs_corrs = [np.abs(a) for a in all_corrs]
   2391     corr_polarities = [np.sign(a) for a in all_corrs]

/anaconda3/envs/mne_current/lib/python3.6/site-packages/mne/preprocessing/ica.py in <listcomp>(.0)
   2387 def _find_max_corrs(all_maps, target, threshold):
   2388     """Compute correlations between template and target components."""
-> 2389     all_corrs = [compute_corr(target, subj.T) for subj in all_maps]
   2390     abs_corrs = [np.abs(a) for a in all_corrs]
   2391     corr_polarities = [np.sign(a) for a in all_corrs]

/anaconda3/envs/mne_current/lib/python3.6/site-packages/mne/utils/numerics.py in compute_corr(x, y)
    245     """Compute pearson correlations between a vector and a matrix."""
    246     if len(x) == 0 or len(y) == 0:
--> 247         raise ValueError('x or y has zero length')
    248     X = np.array(x, float)
    249     Y = np.array(y, float)

ValueError: x or y has zero length

Allison C. Nugent, PhD
Director, MEG Core Facility
NIMH/NIH/DHHS
10 Center Drive, MSC 1059
Building 10 Room 4N242
Bethesda, MD 20892-1059
301-451-8863
nugenta@nih.gov<mailto:nugenta@nih.gov>

PR #6900 (2019-10-11): MRG, FIX: Corrmap check
Closes #6836.

@nugenta can you see if this gives you a more informative error now?
closed 2019-10-11 by larsoner