mne-tools/mne-python #6837: BUG: morph.as_volume does not check subject `src`

larsoner, 2019-09-27:
@jhouck is able to do this for `inv` defined for `sample`:
```
stc_fsaverage.as_volume(inverse_operator['src'], dest='mri', mri_resolution=False, format='nifti1')
```
Assuming the `inv` has a properly defined `src[0]['subject_his_id']` this should not be possible. If that field is not defined, we should probably update the `sample` dataset.

jhouck, 2019-09-27:
I did double-check, and for that file (sample_audvis-meg-vol-7-meg-inv.fif) `inverse_operator['src'][0]['subject_his_id']` is None.

larsoner, 2019-09-27:
Ahh okay. So maybe it does check. But we should probably update the `sample` dataset.

jhouck, 2019-09-27:
Maybe also a warning if you load/create an inverse operator when its subject is None?

larsoner, 2019-09-28:
That's probably a reasonable thing to do, too.

agramfort, 2020-01-05:
agreed. We should have a warning and update the sample dataset.

larsoner, 2020-01-06:
FYI I just tried recreating the source space with MNE-C and it looks like it does not actually write `subject_his_id` for volume source spaces. I'll set it manually, save the file, and update the dataset. (Our Python `setup_volume_source_space` does set `subject_his_id` properly.)

PR #7177 (2020-01-07): MRG, FIX: Check for subject_his_id
Updates `sample` dataset to have `subject_his_id` set in the volume source space. Also adds a `warn` when you give a `src` that does not have it defined.

Closes #6837.
closed 2020-01-07 by agramfort