mne-tools/mne-python #6841: Loading edf raw file: could not convert string to float:

caviri, 2019-09-30:
Hello, I found this small problem when I try to load an edf file

#### Describe the bug
When I try to load an EDF file this error appears:

#### Steps and/or code to reproduce

```Python
import numpy as np
import mne

edf = mne.io.read_raw_edf('/nas-fbm-av1/Roberta/for Carlos-Claudia/E010_RackA_KA', annot=None, verbose=True)
edf
```

#### Expected results
I expect the file loaded

#### Actual results

```
Extracting EDF parameters from /nas-fbm-av1/Roberta/for Carlos-Claudia/E010_RackA_KA...
 file detected
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-13-c29995e686f9> in <module>
      2 import mne
      3 
----> 4 edf = mne.io.read_raw_edf('/nas-fbm-av1/Roberta/for Carlos-Claudia/E010_RackA_KA', annot=None, verbose=True)
      5 edf
      6 

/opt/anaconda3/lib/python3.7/site-packages/mne/io/edf/edf.py in read_raw_edf(input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)
   1237     return RawEDF(input_fname=input_fname, montage=montage, eog=eog, misc=misc,
   1238                   stim_channel=stim_channel, annot=annot, annotmap=annotmap,
-> 1239                   exclude=exclude, preload=preload, verbose=verbose)

/opt/anaconda3/lib/python3.7/site-packages/mne/io/edf/edf.py in __init__(self, input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)

/opt/anaconda3/lib/python3.7/site-packages/mne/utils.py in verbose(function, *args, **kwargs)
    726         # set it back if we get an exception
    727         with use_log_level(verbose_level):
--> 728             return function(*args, **kwargs)
    729     return function(*args, **kwargs)
    730 

/opt/anaconda3/lib/python3.7/site-packages/mne/io/edf/edf.py in __init__(self, input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)
    156         input_fname = os.path.abspath(input_fname)
    157         info, edf_info = _get_info(input_fname, stim_channel, annot,
--> 158                                    annotmap, eog, misc, exclude, preload)
    159         logger.info('Creating raw.info structure...')
    160         _check_update_montage(info, montage)

/opt/anaconda3/lib/python3.7/site-packages/mne/io/edf/edf.py in _get_info(fname, stim_channel, annot, annotmap, eog, misc, exclude, preload)
    400         if annot is not None:
    401             warn('Annotations not yet supported for GDF files.')
--> 402         edf_info = _read_gdf_header(fname, stim_channel, exclude)
    403         if 'stim_data' not in edf_info and stim_channel == 'auto':
    404             stim_channel = None  # Cannot construct stim channel.

/opt/anaconda3/lib/python3.7/site-packages/mne/io/edf/edf.py in _read_gdf_header(fname, stim_channel, exclude)
    700 
    701         edf_info['type'] = edf_info['subtype'] = version[:3]
--> 702         edf_info['number'] = float(version[4:])
    703 
    704         # GDF 1.x

ValueError: could not convert string to float: 
```

#### Additional information
```

Platform:      Linux-4.15.0-47-generic-x86_64-with-debian-buster-sid
Python:        3.7.3 (default, Mar 27 2019, 22:11:17)  [GCC 7.3.0]
Executable:    /opt/anaconda3/bin/python
CPU:           x86_64: 48 cores
Memory:        754.5 GB

mne:           0.16.2
numpy:         1.16.4 {blas=openblas, lapack=openblas}
scipy:         1.3.1
matplotlib:    3.1.1 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.21.3
nibabel:       2.5.1
mayavi:        Not found
pycuda:        Not found
skcuda:        Not found
pandas:        0.24.2
```

Thanks!

cbrnr, 2019-09-30:
It looks like this file is recognized as a GDF file. Could you share the file so that I can reproduce the error?

caviri, 2019-09-30:
Unfortunately I'm not allowed to share the file but thanks for your fast comment. I rename the file with .edf extension and it works.

closed 2019-09-30 by caviri