mne-tools/mne-python #6842: mne.Epochs with event_repeated parameter does throws RuntimeError

nschawor, 2019-09-30:
MNE 0.19 has a convenient event_repeated parameter in mne.Epochs. But it does not work, a RuntimeError is thrown when trying to use it.

#### Steps and/or code to reproduce
```Python
import mne
import numpy as np

nr_samples = 10000
nr_channels = 20
ch_names = ['chan%i' %i for i in range(nr_channels)]
info = mne.create_info(ch_names=ch_names, sfreq=1000)
data = np.zeros((nr_channels,nr_samples))
raw = mne.io.RawArray(data,info)

events = np.array([[1000, 0, 1],[1000, 0, 2]])
epo = mne.Epochs(raw,events,event_repeated='drop')
```

#### Expected results
An epoch structure should be created with the repeated events dropped.

#### Actual results
`RuntimeError: Event time samples were not unique. Consider setting the event_repeated parameter."`

larsoner, 2019-09-30:
@sappelhoff can you look?

sappelhoff, 2019-09-30:
I am on vacation right now with only my smartphone ... So it'll take 14 more days until i can look, sorry-------- Original Message --------Subject: Re: [mne-tools/mne-python] mne.Epochs with event_repeated parameter throws RuntimeError (#6842)From: Eric Larson To: mne-tools/mne-python CC: Stefan Appelhoff ,Mention @sappelhoff can you look?

agramfort, 2019-09-30:
someone can look here at the mne sprint

fmamashli, 2019-09-30:
Ok, I will work on it :)

PR #6871 (2019-10-01): #6842: mne.Epochs with event_repeated parameter throws RuntimeError
#### Reference issue
Example: Fixes #6842 

#### What does this implement/fix?
selection and drop_log should have been included in the _handle_event_repeated function. 
There are now two more input. I also added these new inputs to the tests_epochs.py. It still need to be fixed. 

CC: @agramfort please take over ...
closed 2019-10-01 by larsoner